maelstrom2001/ArcticFox #3: TC algorithms

kablek, 2016-12-18:
I really like the way myevic has algo settings, is there anything similar coming up here? Maybe you could even enhance it and add some sort of a PID autotune option. Default algo seems a bit clunky to me.

spado92, 2016-12-18:
+1 to this!
I was coming here to suggest just the same thing.
The commit on myevic related to such PID can be found here: https://github.com/ClockSelect/myevic/commit/7595147afac067abf4507deb31fd4a21289ba0f4
This one is the one related to all the other algos and their menu implementation:
https://github.com/ClockSelect/myevic/commit/ab2c0405d20e5b352d244efcfd8d566b8f5c8a62

TBXin, 2016-12-18:
Guys, can you explain me what this algorithms give you?
I think this is all placebo... 
I do not believe that they taste better, or something like that.

maelstrom2001, 2016-12-18:
Thanks for suggestion. Currently have no plans to add alternative algorithms.

spado92, 2016-12-18:
Well, I think it has to do with the behavior when the protection kicks in and tames the vape down.
It is particularly evident when you set the wattage way above the needed one and with the current algorithm it will jump up and down making a clear hissing sound as well.
In the coming days I will try documenting it using the monitoring function and will follow up with`more details.
For now, thanks for your great work NFE Team!`

kablek, 2016-12-18:
TBXin: Algorithms (in myevic) are methods to determine how much power to put into the coil to maintain set temperature. The basic principle is easy to understand, if coil is to cold apply more power, if it's too hot reduce power, but the trick is - how to determine the correct amount of power?

With your defaul algo for TC I get oscillations in temperature +-10°C at my current setup, some can be worse some can be better, having ability to chose what method of regulation works the best really stabilizes the temperature and prevents overshooting and undershooting.

With quick test on my evic, I could get the overshoot/undershoot to +-4°C which is significantly less oscillation than with arcticfox, tbh I think algo setting is the only real advantage of myevic over arctic fox.

It's not about tasting better really, but right algo can clear up tanks that sometimes spit, it can be tuned to reach temperature faster. There is no best algo tho, it all depends on what kind of atomizer/coil you are using.

TBXin, 2016-12-18:
@kablek did you try TC with TFR materials in ArcticFox? 

P.S.how this oscillations impacts your vape experience? Just more smooth curves in the device monitor?
P.S.S To reach temperature faster you can also use preheat or power curves.

kablek, 2016-12-18:
I did try both, I obviously prefer TFR it gives me much more stability and better repeatability, I know it is better (I am electronics engineer student so I know the actual difference) but that is not really the source of oscillations.

Oscillations are purely result of regulation. And it is not just smooth curve on monitor, sometimes I can notice that drag is uneven, because of oscillations it seems like the vape is pulsing the temperature so you get short shootups of warm vape and then feels like there is nothing coming out. This effect is sometimes worse sometimes better, it really shows up once you have batteries that are not fully charged but I know it still has enough power to supply me stable flow of vapor.

dywersant, 2016-12-25:
Its a needed feature for devices with very bad and noisy resistance measurement, like all the rx200.
Here are some samples, believe me there was huge difference in vape quality,
[lnk](https://github.com/ClockSelect/myevic/issues/110#issuecomment-257140916)

spado92, 2016-12-26:
Here is my current setup on a VTwo Mini HW 1.01:
Arctic Fox
[image]
Myevic with PID setting
[image]

Vinna152, 2016-12-28:
@TBXin I believe you'll only notice a difference in flavour when tasting with complex juices.

When You have a complex juice different flavours become more pronounced at different temperatures.

So having a curve starting at say 75% and ramping up to 100% will reveal the subtler notes of the flavour as well as the main notes.

TBXin, 2016-12-28:
@Vinna152 i don't use TC at all. Only VW, only hardcore. My vape experience is too different from experience of people that are using TC. E.g. 110 watts, complex builds, like stapples or tripple staggered claptons.

TBXin, 2017-01-03:
Done in the [ArcticFox 170103](https://github.com/maelstrom2001/ArcticFox/releases/tag/170103).

We added configurable PI Regulation for TC modes.

closed 2017-01-03 by TBXin

e30Birdy, 2017-01-31:
What is this PI regulation?

TBXin, 2017-01-31:
@e30Birdy https://github.com/maelstrom2001/ArcticFox/issues/91