maelstrom2001/ArcticFox #4: Battery voltage RX200

frogontherox, 2016-12-21:
The Battery Voltage on the RX200 HW Ver.1.00 is going Wrong. The batterie are fully loaded @4,17V.

[image]

TBXin, 2016-12-21:
Thanks for very fast report ^^
It seems different HW versions used different DAC's.
We are working to fix this issue on 1.00, 1.03 hardwares, 1.04 works fine.

frogontherox, 2016-12-21:
Thank you for the answer. The Arctic fox is a great firmware!!!! I hope it will work on my "old" RX200.
I use this mod on my other divices (Evic Vtc, Cuboid, RX 2/3) and it works great! Thank you.

TBXin, 2016-12-21:
It will! We just found what causing this strange behavior. Today we will create ArcticFox release with fix.

frogontherox, 2016-12-21:
Wow! I will test it ;-) Thank you!

maelstrom2001, 2016-12-22:
Please test the latest release: https://github.com/maelstrom2001/ArcticFox/releases/161222

frogontherox, 2016-12-22:
Tested! Works great!

maelstrom2001, 2016-12-22:
Thank you for response, nice to hear that bug was fixed :)

closed 2016-12-22 by maelstrom2001