maelstrom2001/ArcticFox #7: Setting Clock

meehr, 2016-12-23:
Hi, i dont find Setting Options for Clock. 

I hope you can help me. 

Thanks and Happy Christmas

maelstrom2001, 2016-12-23:
Hello! Thank you, and Happy Christmas to you too :)
Main Menu -> Screen -> Clock for change style and behavior.
Main Menu -> Settings ->RTC for setup RTC module.
Also, note that you can't choose "Anaog" option on 96x16 screens, there is "Digital" only and this menu item is invisible.

vladxed, 2016-12-25:
is there a way to set it up to 12-hour time format?

maelstrom2001, 2016-12-25:
No, 24h format only.

closed 2016-12-26 by TBXin

inabsentia-jacob, 2017-03-05:
Sorry for necroing an older issue like this, but I'm having a hard time setting up the "RTC adjust" and "RTC speed" parameters correctly on my iStick TC 200W. "RTC speed" is set to 33633, which is what it defaulted to when I first installed Arctic Fox version 170201, but I've noticed that the clock is running fast by roughly 3/4 minutes per 24 hour period.

Perhaps you know what the correct setting would be; if I understand correctly what I've been reading on the subject, it relies on some sort of crystal present inside the device? If that's correct then maybe there's a known accurate RTC speed for this particular mod. 

I'm also unable to work out the function and correct use of the "RTC adjust" parameter and would appreciate if you could explain it somehow.

Thanks in advance for your help.

TBXin, 2017-03-06:
@inabsentia-jacob 

Hello,

Unfortunately there are no "correct" speed for LIRC. 
Each individual instance has its own unique rate, so even having two of the same devices - they will have different rate.

You can find best value for you only trying again and again, until the result will satisfy you.

If clock running to fast - decrement value, if slow - increment.
Or you can just enable "LSL" option and get almost perfectly running clock, but it will increase power consumption a bit.