maelstrom2001/ArcticFox #11: [bug] Wrong resistance reading on cuboid

kablek, 2016-12-24:
I have recently bought used joyetech cuboid (150W fw upgradeable to 200W) I can not confirm if it is just my unit, but I have read on forums that a lot of people have the same issue! It seems my cuboid (HW rev. 1.01, ArcticFox 161222) reads resistance higher than my other device (joyetech vTwo mini - which works great) - no mater which firmware I put on it reads high (even official). But I have feeling it might not just be my device but whole series of them? I need more people to confirm that! I have seen a lot of people have similar problems on official fw and cuboids... Now this will be long message since I want to describe problem as good as I can.

In the test first coil is what I had built first time I noticed problem, so I no longer have first coil, second coil is what I currently use.

-Vtwo reads approx. 0.650ohm and on another coil 0.540ohm
-cuboid reads approx. 0.9 - 1ohm and on another coil 0.7ohm-0,9ohm
(both coils SS316L)

I have checked inside the unit (nothing looks damaged) and compared what I saw with some other units disassembled on line. What got my attention is shunt resistor, from disassembly of official firmwares and from source code of myevic firmware I have noticed that shunt resistance on cuboid (resistor for measuring current) should be 1.05mOhm (I assume AF uses the same value). I also found out that ADC reference voltage should be 2.56V. The voltage on shunt seems to be amplified with something that looks like INA199A2 and other people on some forums said that this is the chip commonly used for that task on evic stile boards.

The shunt resistor in mine however is marked 071 and has an underline (underline is commonly used for current sense resistor to indicate it's miliohms) however there is also a bit more space between 0 and 7 and I am not quite sure if there is no dot between them (because of underline) assuming this markings are the resistance value it would be 71mOhms, that is waaaay too high of course and since spacing seems a bit weird we can assume it would be 0.71mOhm.

First I will assume we are putting 25A through shunt resistor (since people complain it is the hard limit for current on their cuboids) and we do the math (electronics engineering stuff bellow).

-First for what it should be (1,05mOhm)
U=R*I=0,00105*25=0,02625 ->we amplify with INA199A2 gain (100V/V) ->2,625V
that is just a bit over what it should be (reference voltage for ADC-2,56V we are 0,065V above that)
so when we turn the equation around we get that max measure-able current is 24,38A this seems okaayish

-Now with what seems to be inside mine (0.71mOhm)
U=R*I=0,00071*25=0,01775 ->amplify with (100V/V) -> 1,775V
we see there is a bit more headroom now since we are in not over 2,56V.
Again we turn the equation around and get max current we can measure:36,05A ->gives me suspicion they wanted lower resistance so they can measure higher currents but screwed up some logistics/forgot they did that and left official firmware on what it was since it has same problem?

okay so I grabbed multimeter and measured what mods are outputting when there is no atomizer present (reading resistance works the way it pushes know voltage on atomizer and then it measures current) (I measured over 10kohm resistance to mitigate cap charge building up voltage)
Ucoilidle(vtwo)=0,5V
Ucoilidle(cuboid)=0,5V
So for first coil:
I=0,769A, Ushunt(1,05)*gain=0,0807V, Ushunt(0,71)*gain=0,0546V
and for second one:
I=0,926A, Ushunt(1,05)*gain=0,09723V, Ushunt(071)*gain=0,0657V

R=U/I
I=Ushunt*Rshunt

so lets assume we compare right calc vs wrong

for first coil:
R(right)=0.5V/((0,0807V/100)/0,00105)=0,651ohm
R(wrong)=0.5V/((0,0546V/100)/0,00105)=0,962ohm

for second coil:
R(right)=0.5V/((0,09723V/100)/0,00105)=0,54ohm
R(wrong)=0.5V/((0,0657V/100)/0,00105)=0,799ohm

soooo it seems there is something about that resistor. Buuuuut that is not the end of the story yet.... I have noticed that if I leave my cuboid idle, resistance slowly creeps up (in case of second coil to about 0.95-1.04 ohm. And as soon as I fire at preety much anything above 1.5V (about 1-25W keeps resistance the same). So it really is confusing, at 5W (that pretty much still keeps coil cold, absolutely no vapor production) resistance reads at 0,7ohm.

I am wondering if they got some shit resistors into production or wtf is going on and if it's just my unit. I need more cuboid users to confirm this.

kablek, 2016-12-24:
small update - I connected 0.33ohm 10% resistor to both mods to get reference measurement, Vtwo was almost spot on at 0.33-0.34ohm, cuboid however was set at 85% shunt compensation and was showing from 0,37-0,47ohm, however, when I fired the mod through resistor it was showing exactly 0.33ohm...

dywersant, 2016-12-25:
Got the same problem on mine. Hw 1.00, one of the first batches. Mine is reading around 10-20% resistance above what it should. I had a look at the pcb but havent been able to spot any problems, 510 is in good shape.

TBXin, 2016-12-25:
@dywersant you can try to fix this using SHUNT correction in the Menu -> Settings -> Expert -> SHUNT, set value to 90%. This will reduce resistance value by 10%

kablek, 2016-12-25:
@dywersant can you check the PCB again and see what the shunt resistor is? Since this got marked as "wontfix" I decided to go with compiling my own source and debugging in myevic firmware since this ones source is not available. I have not yet completely figured out how, but I have mitigated one problem (resistance dropping after fireing and then slowly creeping back) however resistance reading is still wrong, but with some adjustments to TCR I managed to get TC working on SS316L (with TCR value of like 60)

dywersant, 2016-12-25:
Which part is that? Sadly, I am not familiar with electronic components.

kablek, 2016-12-25:
picture from this video: [youtube](https://www.youtube.com/watch?v=jOi_MMnl-Qg) (8:09)

its R27

[image]

This is the resistor used for measuring current.
On video it is underlined 02 which doesn't make sense since that would be 0,2 mOhm which is very low for measuring current (but it can be just code, sometimes digits represent actual value sometimes manufacturer part num etc.).

on mine it says 071 (underlined ->underline is important, that would make it 0.71mOhm which makes sense, but we can't just confirm that, I could measure it but I don't have anything to measure such low resistance with, fact is that setting resistance value in firmware to 0.71mOhm gets me far more acurate reading than default 1.05mOhm.

kablek, 2016-12-25:
okaaaaj sooo I hooked voltage/amp meter to my cuboid and vtwo and measured the stuf, Vtwo was pretty much spot on, Cuboid however was first wrong with amps, which I calibrated (shunt value of 1mohm seemed to be spot on) tested some different resistances, current was always correct, but then I measured the voltage and it was off by 0.34V (instead of 2.44V it was 2.1V on my multimeter) now I don't know if the error is % and the problem comes from scaling the voltage down or is it software? or even chip?

dywersant, 2016-12-25:
[image]

kablek, 2016-12-25:
okay so now we know version 1.00 and 1.01 potentially have different shunt resistances, how about voltage, do you have way of accurate output voltage measurement?

kablek, 2016-12-25:
@dywersant can you hop into this IRC chatroom so we can check if our problems are identical? [http://webchat.freenode.net/?channels=E_UN&uio=d4](http://webchat.freenode.net/?channels=E_UN&uio=d4)

Jsl-1, 2016-12-26:
Hi, dont know if it's the same issue, but I need the same shunt correction (85%) on egrip 2. No correction is needed on my istick pico.

Thanks for all your work on this amazing firmware! And thanks to make  it working on istick pico ;)

kablek, 2016-12-26:
okay, so I calibrated my version of myevic firmware to use - 1.00mOhm shunt resistance and to scale down al the voltage readings by 20% (because it was a bit wrong) now resistance reading seems correct, but resistance still creeps up when you don't fire it. I then raised the voltage cuboid uses to probe and it really reduced that effect too! 

I am really starting to get a grip of this vape programing and firmware building, so if you guys need developer and tester for cuboid here I am. :)

dywersant, 2016-12-26:
I dont have a multimeter.  Whats the point anyway since the bord is reading resistance incorrectly.

kablek, 2016-12-26:
Point is that it is reading resistance by applying voltage to atomizer and measuring current, It needs to measure both current and voltage correctly to read resistance right. And on my unit both current and voltage was read wrong.

kablek, 2016-12-26:
I actually went through the trouble to measure out what vape is outputing/reading and what it should read with my meter. I made some fancy graphs, I set shunt to 1mOhm and started measuring. ->coil resistance is 0.66ohm (actuall fixed resistor, tolerance 5%)

[image]

we can see, current is quite precise in fact, except first reading at 1W. It looks like it is fluctuating a lot on graph but it is not, it is between 102-104% pretty much all the time, note that current error was calculated from resistance and voltage so it might not be 100% actual graph, I expected this graph to slowly rise because of higher power being pushed into resistor more heat is generated/resistance rises a bit.

what is interesting is Voltage error, that was directly measured. We see we got quite a lot of noise at lower power, but it seems to clear up at 20-30 watt. Problem is I don't really see a mathematical way to compensate for that an I have no idea how repeatable this result is.

dywersant, 2016-12-26:
Any idea which part is malfunctioning and needs to ne replaced?  At some point in the past it was functioning properly,  so something got broken i guess.

kablek, 2016-12-26:
I'm not sure if anything is really broken or is it designed badly from the firs place I am struggling to compensate stuff now to see if I can make it work.

kablek, 2016-12-26:
Okay, so I have made my own version of myevic firmware where I compensated voltage error a bit (not fully but it's much closer) you can try it on your vape and tell me if TC works.

https://www.dropbox.com/s/phdpbkc1hs7cyfj/myevic.bin?dl=0

Note that this firmware should not be used for everyday since I had to ramp up probing voltage to get somewhat useful data. Note that it is still not 100% precise but it is now able to handle TC. If it works for your unit too then it is probably nothing broken in hardware, but it's something in firmware.

maelstrom2001, 2016-12-26:
> version of myevic firmware

Do not you think that will be more logical to continue the discussion in the **myevic** repository, if you make version of **myevic**? Not ArcticFox? :)

kablek, 2016-12-26:
sorry about that, unfortunately I don't have access to ArcticFox source, but my guess is both firmwares use same code for part that is causing trouble since both have same problem. Above firmware I compiled is just for reference to see if it fixes the same problem also on other units. Would love to fix the problem in arcticfox tho since it is much better imo.

Gremdude, 2017-01-01:
So with all this talk about boards and chips, this problem is fixed with a firmware patch? The reason the moderator said this wasn't an issue is because you made it seem to be a hardware issue! I am having the same problem if the Cuboid reading resistances way off. Also, the resistance lock doesn't work with ArcticFox on my Cuboid

kablek, 2017-01-01:
it actually is a hardware issue that can kind of be patched in software :/