maelstrom2001/ArcticFox #8: [RFC] Auto disable usb charging when board heated more than 65°C

Nefelim4ag, 2016-12-24:
AFAIK RX series have a problem when usb charging is used (i don't know about another devices).
I use it sometimes, and i think not only i'm.
But i afraid of destroy my board by using usb port for charging =\

What about add logic to check board temp and stop usb charging when temp is higher then threshold(65°C as example)?

May be this can be useful?

Thanks.

TBXin, 2016-12-24:
Do not charge RX via USB and you will be fine. Also, you can disable charging in the options.

[image]

Nefelim4ag, 2016-12-24:
I know that i can disable USB charging manually and as i read on forums RX2/3 after several month of charging continue working good, i.e. i just think about it for be on the safe side.

traviswparker, 2017-01-26:
FYI, the stock firmware for the RX200 will display "Charge Error" and stop charging when the board hits 65ºC. The cutoff might actually be in hardware because it takes a while to reset. Charging will work properly once the board cools down.

maelstrom2001, 2017-01-29:
Done in nightly.

closed 2017-01-29 by maelstrom2001