maelstrom2001/ArcticFox #14: Setting Clock

anfichtn, 2016-12-26:
Specially Devices without backup battery...  Is there a way to update the date and time informations with any android app by only connecting usb to the phone?

maelstrom2001, 2016-12-26:
No. Date/time autosync is available with PC running NFE Toolbox only.

MoDuLah, 2016-12-26:
Simply put the device on a usb power supply while changing battery/ies ;)

maelstrom2001, 2016-12-26:
"Great" advice for mods with 2 or more batteries.

MoDuLah, 2016-12-26:
With disabled charging it just provides enough power to don't lose the time settings. ^_^

TBXin, 2016-12-26:
You can damage your device this way, also, if the atomizer is plugged this will lead to the constant on / off board.

maelstrom2001, 2016-12-26:
If somebody forgot to disable this option, he will have to buy new Cuboid. So, please...

closed 2016-12-26 by maelstrom2001

JohnW383, 2017-01-04:
I found out tha with my rx200, if I change the back 2 batteries 1 at a time, take out the left leaving the right one in, put new one in left then change the right, the clock saves the time. I'm just trying to get the clock speed right, it keeps running fast, but I'm playing with the speed and it's getting closer.