maelstrom2001/ArcticFox #16: mode button usage?

AAI688, 2016-12-29:
Hi Maelstrom,

Thank you for this firmware as is breathes new life for my ipower and istick 100w tc.. its just fantastic with the new features.. Just wondering as a suggestion for those 2 mods i use, they have a mode change button which in the orig firmware is used to change modes.. atm i believe they are not being used in any ways? is there a way to config it like in the myevic (button clicks) to be used for other purposes? That would be cool! I would be happy to test them out.

Thanks In Advance!!

maelstrom2001, 2016-12-29:
Thanks for suggestion. Maybe, a bit later :) We have to do more other things right now.

maelstrom2001, 2017-01-22:
Done in nightly builds. Hold Mode button for about 1 sec to select profile.

closed 2017-01-22 by maelstrom2001