maelstrom2001/ArcticFox #19: Joyetech Cubis QCS coil in TC mode

vendeghlajos, 2016-12-30:
Constant error message, only power mode is usable.
0.5 Ohm SS coil is working with TC without any issue.

maelstrom2001, 2016-12-30:
Try to disable TCR check: Main Menu -> Settings -> Expert -> ChkTCR set to Off.

vendeghlajos, 2017-01-01:
same with TC check OFF

closed 2017-01-01 by vendeghlajos

TBXin, 2017-01-01:
DId you know from what material this notch coils are made?

vendeghlajos, 2017-01-01:
SS316

TBXin, 2017-01-01:
Any way, you can try custom TCR: 92 with this coils.

vendeghlajos, 2017-01-01:
It is basically same as the Joyetech Notch coil. 

http://www.joyetech.com/product/notchcoiltm-0-25ohm-dl-head/

TBXin, 2017-01-01:
> It is basically same as the Joyetech Notch coil.

You can try TFR 316L or custom TCR: 92 with this coils.

vendeghlajos, 2017-01-02:
same error message on both. sorry to say.

TBXin, 2017-01-02:
@vendeghlajos okay, this coils work properly with OFW ?