maelstrom2001/ArcticFox #20: Locked resistance NOT locking

Gremdude, 2017-01-01:
Good afternoon and happy new year! With the ArcticFix firmware on my Cuboid, the locked resistance does not work. The Cuboid reads the resistance way off with a .11 ohm coil reading .15 ohms. I use the profile screen to adjust the ohms down to the correct value and press exit. The resistance just stays at .15. If I do this a couple times, it will eventually retain the corrected value but as I Vape a few times, the profile selection screen comes up. I select the correct Nickel profile with the locked .11 ohms showing. When I press fire? Back to .15 ohms. I was previously using the Myevic firmware and did not experience this issue. I like your firmware better because of the NFE TOOLBOX connectivity but since I am a temp control guy, this issue is a killer! Thanks for your time!

maelstrom2001, 2017-01-01:
Turn off Smart option. Every time Cuboid wakes up, it re-read atomizer and can not find right profile: 0.15 ohms is too high to be detected as 0.11 ohms locked. But if Smart will be off it will accept 0.15 ohms as 0.11 locked. Also, edited resistance is displayed on screen only on first firing after editing, by design.

Gremdude, 2017-01-01:
Thank you. That worked! That being said, on the Myevic and stock firmware Smart mode is not an option for temp control. Perhaps it might be a good idea to automatically turn off Smart when using TC? Just a suggestion! Thanks for the assutamce and keep up the great work!

Gremdude, 2017-01-01:
I meant assistance! Can't edit posts once they are posted I guess...

maelstrom2001, 2017-01-01:
> Perhaps it might be a good idea to automatically turn off Smart when using TC?

To be honest, I never got the same glitches. Maybe because I use lightweight coils in TC, such as titanium/SS316 4..5 wraps of 28 AWG on genisis atomizers, and its resistance while mod is wake up is always in default Smart range (10%). Anyway, glad to know that now is OK :)

closed 2017-01-01 by maelstrom2001