maelstrom2001/ArcticFox #21: Puff counter digits number

vendeghlajos, 2017-01-01:
When not needed, it could be only 1-2-3 digit visible without the first zeros. I mean 00200 would be displayed as 200. Thx.

TBXin, 2017-01-03:
Done in the [ArcticFox 170103](https://github.com/maelstrom2001/ArcticFox/releases/tag/170103).

You can switch between two styles: seconds (without zeros) and H:MM:SS.
To change style you should use [NFE Toolbox](https://github.com/TBXin/NFirmwareEditor/releases/tag/6.4).

closed 2017-01-03 by TBXin

vendeghlajos, 2017-01-03:
sorry, but my idea was not about the time, but about the puff numbers. probably you closed this accidentally.

TBXin, 2017-01-03:
I don't think so ^^

[image]

vendeghlajos, 2017-01-03:
achhhh, i updated the mod. thank you NICE!!!