maelstrom2001/ArcticFox #22: puff counter auto reset - every morning

vendeghlajos, 2017-01-03:
Hello, 
In expert mode, I switched on the RCOBC.
Randomly I get the possibility to reset the puff counter on the main screen.(keep - reset option is there)
can you explain, how this works? It is there any key combination to get into the puff counter reset?
RECOMMENDATION: I am using the counter for daily tracking, would be perfect, some auto-reset option, for example, every morning at 6:00 a.m.
thx!
Lajos

TBXin, 2017-01-03:
Hello,

RCOBC means Reset Counters On Battery Change.
Dialog will appear when you device is turned on after you insert battery (every cold startup).

vendeghlajos, 2017-01-03:
thx, now it is clear. Probably I disconnecting the battery time to time.
Could you consider in the next releases about the daily auto reset option?

TBXin, 2017-01-03:
If you want to reset the counters - do it manually.
We will not implement automatic actions, tied to the time.

vendeghlajos, 2017-01-03:
Sad to hear...I think many users would appreciate this option, and no firmware offer this option.
OTHER OPTION: most of the users are fully charging his devices during the night - when 100% battery level reached, after unplugging from the charger could be offered the option of the auto reset.
I do not see the point, why after the battery disconnection is the reset option offered. Many devices has a built-in battery.

blitzu, 2017-01-03:
I want this function too

febs, 2017-01-10:
It would also amazing to have separate puff counts per profile.
This way I could count how many puffs I had on my Nautilus and how many on my TFV8.

maelstrom2001, 2017-01-10:
@febs 
> separate puff counts per profile

Sorry, it's cool feature but it will be just wasting of EEPROM space and, which is more critical - its lifetime. Nothing personal, just the reality :)

vendeghlajos, 2017-01-10:
but the possibility of  the "morning auto reset" or the " auto reset after the battery is fully charged" feature you found acceptable? ;-)

maelstrom2001, 2017-01-10:
@vendeghlajos At at this moment can not say exactly when it will be implemented, we have a huge todo list

vendeghlajos, 2017-01-10:
Thank you!
PS: would be good to share with us your plans - what we could expect in the exciting future!

TBXin, 2017-01-10:
You can follow our twitter account to get informed about updates and about announcements.
https://twitter.com/NfeTeam

closed 2017-01-17 by TBXin