maelstrom2001/ArcticFox #23: Possibility of supporting Elitar?

AAI688, 2017-01-04:
Hi Maelstrom,

Previously i asked in the myevic thread for support on Elitar. at the moment i am liking this firmware more due to the profiles. but u guys are doing a great jobs on both sites. sorry if i am abit impatient, any possibilities of supporting Elitar on this firmware in the future? anyways Thanks alot for now. greatly enjoyed this on my other mods that is supported. :)

Cheers!

maelstrom2001, 2017-01-04:
Hello! Thank you. Sure, it is technically possible, but the support of one more screen resolution requires a bunch of space in ROM, so it can be implemented in a separated branch only, and it will take a lot of time.

FrankBKK, 2017-02-09:
Would also highly appreciate that - the original Elitar FW is horrible - still not figured out how to reset the puff counter ...

maelstrom2001, 2017-02-09:
> the original Elitar FW is horrible

Agree that. But at this moment have no plans to buy relatively expencive hardware just for implementing support for 2 boxes with that display resolution: Joyetech Elitar and Eleaf Aster RT. Not very popular devices IMHO.

closed 2017-05-15 by TBXin