maelstrom2001/ArcticFox #25: will you be adding rx300

docdeath, 2017-01-04:
i love your firmware and makes std firmware look stone age  so just asking  thanks

maelstrom2001, 2017-01-05:
Yes, a bit later. Also, we will need the advanced beta testers, as we will announce later.

docdeath, 2017-01-05:
thanks for reply well any time you need beta tester let me know there thanks

closed 2017-01-12 by TBXin

maelstrom2001, 2017-01-13:
Well, added early support of RX300: nightly [af_170112.bin](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170112.bin)
**Untested, may be very buggy, use at your own risk, no warranties at all!**
Use nightly [NToolbox](https://ci.appveyor.com/api/buildjobs/q43eu4e4eylvf107/artifacts/src%2FNFirmwareEditor-v6.4.17.zip) to install and configure.

docdeath, 2017-01-13:
it loaded firmware OK  but its saying chk battery's going in to bvo it only shows one battery  other 3 line all show 0.00