maelstrom2001/ArcticFox #26: RX200S device to hot very fast

Hotix71, 2017-01-05:
Hello!

Iam currently testing your firmware, it looks very great but i've an issue with the RX200S. After doing a couple of vapes at 60w i get a device to hot. It is possible to make the temp control a little bit more flexible (similar to shunt correction or just to set a few degrees more to the overheating protection) , this is a known issue from the past also with other custom firmware running with the RX200x devices.

Thank you!

maelstrom2001, 2017-01-05:
Hello! Please provide information on your setup: we think that maybe DC-DC converter dissipates a lot of heat due to low output voltage? Thanks!

Hotix71, 2017-01-05:
Coil 0,54 Ohm at ~5.6V, what to you need exactly?

maelstrom2001, 2017-01-05:
OK, will set "Too Hot" threshold a little higher in next release.

Hotix71, 2017-01-05:
Great! Thank you very much!

maelstrom2001, 2017-01-22:
Done in nightly. Also you can select temperature sensor in Expert menu, external or internal.

closed 2017-01-22 by maelstrom2001