maelstrom2001/ArcticFox #28: Just in case : trouble with af_170105 with eleag pico mega.

Sidosido, 2017-01-05:
Hello
Upgrading from 103 to 105 (nfe 6.4.2) leads to a semi-bricked box.
Was able to downgrade and recover without any issue to 103 with fire + +.

I upgraded twice, same result : no reaction whatsoever, no screen, no logs, no nothing.
Upgrading succeeds, but in the meantime, the box shuts off completly.

Anyway, tremendous work, keep it up !!!

TBXin, 2017-01-05:
You should use NFE 6.4 or higher to upload ArcticFox 170103+

TBXin, 2017-01-05:
> the box shuts off completly.

Seems to be your devices is stucked in the LDROM boot mode.

Try to upload new firmware and if black screen appears again - reconnect device to PC and look at the "Boot Mode". If it will be "LDROM" just press button, that i pointed in the screenshot.

[image]

maelstrom2001, 2017-01-05:
Also please note that downloads from Dropbox may be corrupted.

Sidosido, 2017-01-05:
Well, I did tried with 6.4.5., same problem again. And I cannot flash once in 105, since no device at all is detected.
Anyway, I can get it back with fire++, so, no trouble at all.

Retried once again with a fresh dl of 105 : I now have a message : "firmware file is not suitable for the connected device."

TBXin, 2017-01-05:
> Retried once again with a fresh dl of 105 : I now have a message : "firmware file is not suitable for the connected device."

This can mean only one thing. File that you've downloaded is broken. Proxy or dropbox issue.

Sidosido, 2017-01-05:
Yup, tried just a few secs ago from another source again, and it works...
Funny I had not the message before.
Sorry for bothering.

closed 2017-01-05 by Sidosido