maelstrom2001/ArcticFox #29: Switch + and - buttons for power with display orientation at Pico

Musik234, 2017-01-05:
There was a question in december about the buttons for the menu being switched when the display is flipped at the IStick Pico.
I think it would be a good idea to switch the buttons also for changing the power; the upper button should increase the power, the lower one should decrease it.

TBXin, 2017-01-05:
For some strange reason Pico OFW has left-hand orientation. If you flip screen via NFE Toolbox to the right-hand - buttons will work as you expected. Upper - increment, lower - decrement.

Musik234, 2017-01-05:
So it's not possible to change this in the firmware?
Is it sort of a "hardware thing", which button increases and which one decreases?

TBXin, 2017-01-05:
It's possible, but it seems to me that Pico users have become accustomed to the current layout of buttons.
and this innovation will not please them, and this innovation will not please them.

Musik234, 2017-01-05:
Just an additional selection in the menu?
"Switch PwrBtn: yes/no"
It's not really an issue for me, just would be nice.

Musik234, 2017-01-05:
I would be happy about it. :-)
Thank you.

Btw: thanks a lot for your great work.

sandbagjohnny, 2017-01-07:
This would be weird.. the buttons are physically marked - and + on the Pico..

Musik234, 2017-01-07:
Using them **intuitively** you don't have to look at the buttons. :-)
Upper one means plus, lower one means minus, no matter what the marks say.

But you are right, that "can" be confusing. Therefore it should be a choice in the menu with the default "plus means increase, minus means decrease".

maelstrom2001, 2017-01-22:
Done in nightly.

closed 2017-01-22 by maelstrom2001