maelstrom2001/ArcticFox #31: функция от myevic

SvEgiiVEteR, 2017-01-06:
в прошивке myevic есть такая полезная штука.
когда паришь в ТК то при выборе настроек предхита(у меня стоит 2 клика) меняется регулировка либо вата либо температры,очень удобно. можно ли и тут такое сделать?

TBXin, 2017-01-06:
Menu -> Profile -> T.Dom 
Or 
Menu -> Settings -> Clicks -> assign T.Dom function to the any available combination.

SvEgiiVEteR, 2017-01-06:
@TBXin 
да это то самое, но у myevic T.Dom и preheat работают на одном режиме(например 2 клика) в зависимости от режима(TC или VW)
это очень удобно. потому что все клики у меня уже заняты....

Yes it is the same, but myevic T. Dom and preheat work in one mode(e.g. 2 clicks) depending on the mode(TC or VW)
it is very convenient. because all the clicks I have already taken...

SvEgiiVEteR, 2017-01-06:
@TBXin ну я месяц юзал myevic.
в режиме VW нажатие 2 клика(у меня там предхит) открывалась настройка предхита,удобно если надо его выключить или настроить.
когда включал ТК на нерже например, эти 2 клика не открывали предхит а меняли режим регулировку: либо ват лимбо температуры. видимо как то можно сделать.. 

а вообще эта прошивка в разы удобнее. меню и настройки более скомпанованы и не разбросаны))
спасибо!)

TBXin, 2017-01-06:
@SvEgiiVEteR под включением ТК вы подразумеваете смену профиля в myEvic?
Если так, то да. Профили в myEvic реализованы совсем не так как у нас. Там профиль это все настройки устройства, в том числе и клики. У нас же профиль это настройки, исключительно связанные с парением: материал, температура, мощность, преднагрев, кривые, но не яркость, клики и т.д.

SvEgiiVEteR, 2017-01-06:
@TBXin да смену профеля с ВВ на ТК(никель сс или титан)
теперь понятно... жаль.
вообщем это единственный по сути минус по сравнению с myEvic.
ну если это как то осуществимо то буду ждать) а так спасибо)

TBXin, 2017-01-06:
@SvEgiiVEteR мы подумаем над этим, возможно поступим так же как с отображением инфо-линий. Для VW один, для TC другие.

maelstrom2001, 2017-01-22:
Done in nightly.

closed 2017-01-22 by maelstrom2001

SvEgiiVEteR, 2017-01-23:
поставил) все ок) спс большое)теперь все очень удобно!)