maelstrom2001/ArcticFox #32: Question: Could you add function to limit max battery drain?

HansWu123, 2017-01-07:
Hi,

could you add a function to limit/set max ampere drain, so that you can set it according to the type of battery you are using?

reason: to account for the worst case scenario during regular mod use
->almost empty batteries, device (accidentally, maybe drunk) set to max output (e.g. 75W), and (accidentally, maybe its in a pocket) constantly on/off firing the device

Example:
I_batt = 1,20 * P / (n * 3)

I_batt  = current drawn
P = wattage setting
n = number of batteries
3 = cut off voltage
1,20 = 20% loss, accounting for mod efficiency not beeing 100%

worst case:
I_batt = 1,2 * 75W / 3V = 30A

This means you should always use a 30A battery.
But most people dont, because eg. Sony VTC5A has with 2600mAh less capacity in comparison to e.g. HG2 with 3000mAh, or 30Q etc...or simply because they dont know what they should use.
Also information is conflicting, people think VTC6 is fit for 30A but in fact its only in conjunction with a 80°C safety temperature cut off according to the specs. 

It would be very cool if you could set an ampere limit, e.g. select HG2 in the profile as used battery type and then have the FW lock the device to 20A/ 50W.

What do you think?

cheers

forkoz, 2017-01-14:
I would love this. I tried TC with SS316L and in VW I can't limit the power at all. As the resistance changes power goes up. I want to limit to 1 or 2 amps or limit voltage and I cannot at all. Even measly 4.3w will climb.

The TC itself isn't super accurate. I can vaporize at 300F which should be below PG's point but I still get a cloud.
[temps.csv.txt](https://github.com/maelstrom2001/ArcticFox/files/706113/temps.csv.txt)

HansWu123, 2017-01-15:
forkoz, in TC you can 4x click fire and then limit power for TC.

What Id love would be the ability to limit the device e.g. Pico to 50W instead of 75W.
I now just read that even VTC5A is not fit for 30A according to its datasheet.
I think I read the authors said they cant limit Amps or something like that, but I know that a original firmware upgrade of the cuboid changed i from 150W to be able to fire with 200W, so it might be possible.

forkoz, 2017-01-19:
limit power in watts which is ambiguous and resistance dependent, not volts. Can't use 5A laptop pulls when VW or TC will bring the load higher out of my control. The firmware can 100% software limit amps. R*V = greater than the limit, stop or don't fire. It won't work the other way but the goal is safety. I have 5, 10, 20 and 25 amp batteries.

These things are already calculated in the memory anyway.

HansWu123, 2017-01-19:
on battery side in a regulated device: watts = volts * amps
its different compared to a unregulated mech mod, there resistance is important

forkoz, 2017-01-19:
ohms law. watts is resistance dependent. http://www.steam-engine.org/batt.asp

keep voltage constant and push resistance up, amps fall. 

for VW:

9 watts = 3v, 3A at 1.0 ohms
9 watts = 4.3v, 2.15A at 2.0 ohms

a subohm (.5) coil pulls over 4 amps at only 9 watts. that's why the push for higher and higher current limits and smaller and smaller coils. After the mod voltage limits are reached its all you can do to get more "power". Either way I don't think they're interested from the label.

HansWu123, 2017-01-19:
use your example in steam engine and look carefully...at battery drain

with 1 ohm
**Atomizer** 
Voltage	3	V
Current	3	A
Power	9	W

**Battery drain**
Voltage	4.2	V
**Current	2.38	A**
Power	10	W

with 2 ohm
Voltage	4.24	V
Current	2.12	A
Power	9	W

Battery drain 
Voltage	4.2	V
**Current	2.38	A**
Power	10	W

when you change resistance battery drain stays the same....

forkoz, 2017-01-19:
i see what you're saying, I forget about input current.

vapinggirl, 2017-02-01:
I agree to HansWu123 (wurst ? ;)
i know many vapers who would love this Feature 

I would call it a User programmable protection ;)

TBXin, 2017-02-01:
At fact we can only reduce battery drain from hardcoded values. 
Some devices like iStick Pico have 25A max limit, RX300 - 50A limit, and this is a hardware limitations.

That's why we think that this feature more useless than useful.

vapinggirl, 2017-02-02:
I think it could be very useful , see HansWu123s Post

A global wattage Limit in expert menue which overrides the normal wattage would be enough i think

sala91, 2017-04-09:
Would this not be easier to just cut of at battery safe voltage and display lower wattages or charge batteries?