maelstrom2001/ArcticFox #34: Feature request: bypass

febs, 2017-01-10:
Hello,
I would love to be able to use the bypass function as with the original firmwares.

Could that be included in the future?
Many thanks.

maelstrom2001, 2017-01-10:
Hello! Maybe, but this option was not planned from beginning, at all.

febs, 2017-01-11:
Thank you for you reply. May I ask what are the reasons why wasn't it planned? Thanks!

TBXin, 2017-01-11:
Because it is useless? :)

Also, how do you imagine bypass mode on RX200(s), RX300, and other 2/3/4 battery devices with a serial batteries connection.

febs, 2017-01-11:
LOL.

Well, on a single battery mod, like, the Pico or the Aster, wouldn't it be
the same of a mechanical mod?
Mech fans swear that it's the best mode ever, the smoothest, and what the
hyperfuck. But I really don't know. I'd love to know more about it. Would
bypass actually bypass as much as possible of the circuitry, or would that
just try to simulate the same behaviour that one would experience with NO
electronics at all in the way?

Many thanks for any clarifications!

TBXin, 2017-01-11:
Of course it will simulate.

If simplify, the algorithm is like this: calculate watts using battery volts and resistance of the coil and recalculate watts each time battery volts is changing.

TBXin, 2017-01-11:
Good mechanical mods can give you so much power... because there are only one limit - your battery.
For example, single mech.mod with Samsung 25R or Sony VTC5 can give you starting boost with ~100Amps, crazy amount of power...

PCBs in our mods, like Pico or Aster have current limit = 25A and they will never give you mech. experience, even with emulation/simulation bypass.

maelstrom2001, 2017-01-11:
>Would bypass actually bypass as much as possible of the circuitry, or would that
just try to simulate the same behaviour that one would experience with NO
electronics at all in the way?

On single battery boxes bypass mode is pretty close to real mech, but the efficiency is really poor, because DC-DC converter still work in circuitry. Even simple MOSFET-driven handmade box will give you more power than bypass. Yes, in bypass mode you'll have all protections, but you'll never had a good experience as **TBXin** wrote above.

febs, 2017-01-11:
Amazing answers. Thank you so much!

closed 2017-01-12 by TBXin