maelstrom2001/ArcticFox #35: Fast battery discharge with RX200S

PaDo90, 2017-01-10:
Hello,
I noticed that the batteries on the RX200S discharge very quickly with the ArcticFox Firmware.
I think that's not normal ... With the original Wismec firmware the batterie keeps a much longer!

TBXin, 2017-01-10:
1. May be you are using wrong battery profile;
2. May be you are enabled LSL;
3. May be you are keep device connected to the usb, while it doesn't charging;

PaDo90, 2017-01-10:
1. I uses the LG HG2 so i set "LG HG2" in Battery Model
2. The Light Sleep Mode is disabeld
3. No i've not connected the device via USB while i use it ...

maelstrom2001, 2017-01-10:
Try to set GEN battery profile and check voltages below 3,7..3,6V. Use BVO settings to set right voltages, check them with voltmeter, charger, another mod etc. Probably, voltages reads slightly incorrect that not allow to use all batteries capacity.

PaDo90, 2017-01-10:
Okay i've set the Battery profile to "Generic" now i have more % :D 
I  have made a puff and this is the result: http://prntscr.com/du058b

maelstrom2001, 2017-01-10:
Measure voltages on batteries with voltmeter and set BVO to equalize them with real one. I see difference about 0,1 volt, and it can lead to early "Low Battery" message.

PaDo90, 2017-01-10:
I don't have a voltmeter

TBXin, 2017-01-10:
Any external charger with a screen?

PaDo90, 2017-01-10:
Yes the Nitecore D4 Intellicharger ... i will try it tomorrow.
So i should check what my external charger says how many volt the battery have and set the values on BVO? Where i know which is battery 1, 2, 3 ?

TBXin, 2017-01-10:
Tip: 
Fully charge your batteries. Take it off from charger, give they a rest on your table about 10 minutes. Put it back to the charger and look how much volts they have.

For example they have 4.18 volts on each. Put it to the RX200S, open BVO's menu and change BVO's values until bottom lines will not show 4.18 on each battery.

closed 2017-01-12 by TBXin