maelstrom2001/ArcticFox #36: [RFC] Does it possible to calculate battery capasity?

Nefelim4ag, 2017-01-10:
AFAIK this firmware writed for some type of chips in battery mod, may be this chip can monitore current (A) from the battery? If yes, so, can you add some simple calculation battery capacity calculation?
Or may be just calculate efficient battery capacity, by using data from current/voltage sensor for coil (i.e. how many capasity my coil really use on 1 battery charge).
May be it can be simplier to calculate only Watts used.
---
т.к. тут все свои, вопрос сформулирован для англоговорящего населения, но я продублирую на русском.

Это просто идея, т.к. у меня нету внешнего устройства чтобы измерить ёмкость батарей, а ведут они себя порой очень странно (RX2/3), хотелось бы иметь возможность смотреть статистику, сколько mAh с батарей было снято, или сколько ватт было использовано испарителем. Таким образом, если ёмкость в ваттах самих батарей я могу рассчитать с некоторой погрешностью, я смогу увидеть сколько реально девайс использует из этой ёмкости, и мб будет иметь смысл задуматься над покупкой другого комплекта аккумов/второго комплекта.
---
Thanks.

maelstrom2001, 2017-01-11:
Unfortunately, we can't read current drain from battery directly, so it can be calculated very approximately only. Need time to investigate how it can be useful.

В общем, надо пробовать. А пока невозможно сказать, насколько точно получится подсчитать потребляемый ток и, как следствие, емкость, вытаскиваемую из аккумуляторов.

noisexoma, 2017-01-12:
Прикольная фишка, очень было бы интересно проверить емкость своих акков, хотя бы +/- 0,1А/ч.
Если это можно реализовать то я только за!

forkoz, 2017-01-19:
mah consumed from calculations? to go with the total puff time. It would at least compare between measurements on the same device and how inaccurate can it be on single battery devices?