maelstrom2001/ArcticFox #37: [RFC] Use both Ext/Int termal sensors for trigger "device too hot"

Nefelim4ag, 2017-01-11:
Device: RX2/3

In my case Ext have a more high temperature (+5C) then Internal sensor, but i'm not sure it's true for all devices.
I don't know which sensor is a best for triggering "device too hot", but i think, that right direction for that, is using both sensors and if one of they reach threshold (75C), then trigger alarm.

Thanks.

spado92, 2017-01-11:
I think this is related to the 170110 changelog [on nightlies](https://www.dropbox.com/sh/lyk6n0nb4zh6o5y/AAABduZkmH2QUbIkztmZZ59ka/nightly?dl=0).

> 170110
------
[+]     select board temperature sensor, external (thermistor) or internal (in MCU): menu Settings -> Expert -> TEMP Ext/Int

TBXin, 2017-01-11:
> In my case Ext have a more high temperature (+5C) then Internal sensor, but i'm not sure it's true for all devices.
For most devices this is normal behavior.
Thermistor is located near power part of the PCB, which is primary heat generator, when internal sensor is a part of the central processor.

maelstrom2001, 2017-01-11:
This is experimental feature. RX300 have no external thermistor, and temperature monitoring using internal sensor only. Early "Device Too Hot" message is known issue on some RX200 and RX2/3 devices, so maybe this trick with switching sensor channels will help to avoid that problem. On some boxes, again :) 
@Nefelim4ag BTW, is temperature on internal sensor channel always higher than on thermistor? For example, while device in stanby mode or charging, internal sensor return higher temperature than external, but if I do some puffs I can get lower temperature on internal channel.

Nefelim4ag, 2017-01-12:
@maelstrom2001, I check temp on cold device Ext/Int 30/27, so it's always higher.
I don't use device for last 12 hours and get in power on, only now.