maelstrom2001/ArcticFox #38: [Possible bug?] Spikes in live battery voltage reading in device monitor.

sandbagjohnny, 2017-01-11:
Hi, 

Great job on this firmware, it is much nicer than the stock firmware on many levels :)

I'm reporting a possible problem with power over USB. I've got two eLeaf Pico's, bought 4 months apart, the first one in June 2016 and the other in November 2016. 

[Pico A is set to power mode. Voltage spikes are circled in red.](http://imgur.com/T1pUTao)

[Pico B is set to TC PI mode. Voltage spikes are circled in red.](http://imgur.com/81tQnij)

TBXin, 2017-01-11:
This spikes are result of the bad quality ADC and him noise.
Device Monitor just visualize data, that device send.

P.S.
Both repo are our, so you don't need to duplicate issues.

sandbagjohnny, 2017-01-11:
Thanks.

I moved my charging station to another PC this morning as a test. (Sometimes the USB ports lose connection on this PC I'm typing on) I've noticed these spikes in the battery voltage signal went away. Poor signal/connectivity from USB port itself can cause this too? Is this correct?

TBXin, 2017-01-11:
Nope, this is incorrect.

sandbagjohnny, 2017-01-11:
hm, ok. Thank you.. Maybe I just got giddy that I didn't see anything weird on the other PC yet.. I still can't seem to reproduce it on this other different PC though so it's all good.

closed 2017-01-12 by TBXin

sandbagjohnny, 2017-01-12:
Would it be possible to write a noise filter into either the fw code, or the toolbox? Thanks!