maelstrom2001/ArcticFox #39: Question about new batterie profiles

smok-e, 2017-01-11:
I would like to add profiles for some different batteries.
But now I can only use 1 custom profile. 
It would be great if there was an option to save the custom profile (external on my PC)
like the wire profiles. And load a custom profile from a csv file.
Would it be possible to implement this?

TBXin, 2017-01-11:
If i understand you correctly - you want import/export function in the MFE Toolbox - ArcticFox Discharge Profile editor?

smok-e, 2017-01-11:
yes, you understand it correctly. Would that be possible?
This way we could add curves for more batteries.

Joergl100, 2017-01-12:
What discharge current is according to the curves? Maybe...we could build individual ones...with an adequate charger with diagnostics/monitoring as the MC3000?

TBXin, 2017-01-12:
Primary values in the discharge curve is a voltage, not current.

Joergl100, 2017-01-12:
Ok, thanks!

TBXin, 2017-01-15:
Discharge Curve Export/Import will be included in the next release.
[image]

closed 2017-01-22 by maelstrom2001

Joergl100, 2017-01-22:
Excactly the same as unedited in "Custom"...