maelstrom2001/ArcticFox #41: Eleaf Aster... gaining time.

febs, 2017-01-12:
This is likely similar if not identical to this one.

https://github.com/maelstrom2001/ArcticFox/issues/30

That has been closed so I unfortunately have to open a new one.
I did apply the changes recommended in that issue:

Set Expert options:
X32 - Off
LSL - On

But the clock still ran ahead of time. About one hour per day.

Thanks

TBXin, 2017-01-12:
Bad for you :)
Aster does not have RTC module, so talking about accuracy like a joke^^

febs, 2017-01-12:
May I suggest then to disable that for the aster? It makes no sense then.
Just considering.  Thanks for caring!

TBXin, 2017-01-12:
Hide it for yourself.

febs, 2017-01-13:
I did just gave a suggestion, I didn't pretend anything...

closed 2017-01-13 by febs