maelstrom2001/ArcticFox #42: Eleaf Pico Dual support

febs, 2017-01-12:
I just bought it and realized that oh noes Arctic Fox doesn't run on it and oh my godz how much the stock firware suck compared to it.

Any chance?

Thanks!

TBXin, 2017-01-12:
Already done, but not released yet.
You can try nightly build or wait for release.

febs, 2017-01-12:
MARRY ME <3

closed 2017-01-12 by febs

maelstrom2001, 2017-01-13:
Added early support of Pico Dual: nightly [af_170112.bin](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170112.bin)
**Untested, may be very buggy, use at your own risk, no warranties at all!**
Use nightly [NToolbox](https://ci.appveyor.com/api/buildjobs/q43eu4e4eylvf107/artifacts/src%2FNFirmwareEditor-v6.4.17.zip) to install and configure.