maelstrom2001/ArcticFox #46: nightly build wrong (?) battery reading

febs, 2017-01-13:
My battery are showing as fully charged on both my nitecore chargers.
The pico dual with the latest nightly build complains with a warning (!) sign and the batteries look about 25% full in the unit.

Hope this helps.

Thanks.

maelstrom2001, 2017-01-13:
Thanks! Will have a look :)

febs, 2017-01-13:
Hem, by the way, do you maybe have a copy of the stock firmware in the while? I realized it's not available to download LOL :-P and I'd like to restore it as long as the new AF is ready and mature. :-)

febs, 2017-01-13:
> http://www.eleafworld.com/pico-dual
> "Download the file below to put custom logo into your Pico Dual."...

Lol, thanks. I looked in this page:

http://www.eleafworld.com/download/

and it was not there.

maelstrom2001, 2017-01-14:
Please test nightly build on your Pico Dual: [af_170114a.bin](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170114a.bin)

febs, 2017-01-15:
I will as soon as my friend will return it to me! I left it in his car. :-P

mittar, 2017-01-19:
Hi, Pico Dual: af_170114a.bin checked (2xSony VTC6 Fully Charged) - Battery icon is OK, shows fully charged.

maelstrom2001, 2017-01-19:
@mittar Thank you! Please test last nightly: [170119](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170119.bin)
Should work all except Avatar support.

mittar, 2017-01-19:
Not working at all , black display , no reaction on buttons ... back to af_170114a.bin and it is running OK.

maelstrom2001, 2017-01-19:
And what with [170116](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170116.bin) ?

mittar, 2017-01-19:
It is working OK.

maelstrom2001, 2017-01-19:
Thank you! And will this work? [170118a](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170118a.bin)

mittar, 2017-01-19:
Yes, 170118a is OK.

maelstrom2001, 2017-01-19:
Great! Now try again please 170119. If it will not start, disconnect box from PC and open/close battery door. And reflash if box will not start itself. Maybe glitch happened during update procedure.

mittar, 2017-01-19:
Yup 170119 is working now.

maelstrom2001, 2017-01-19:
Thank you very much for testing! :+1:

maelstrom2001, 2017-01-20:
Little update: 170119 removed due to unstable behavior, so please temporary use 170118a(b). Thank you!

mittar, 2017-01-20:
Is the unstable behavior, that ohm reading is at least doubled?
Head 1 (SS316L) - DNA - 0.32, Pico Dual - 0.63)
Head 2 (Tungsten) - DNA - 0.65, Pico Dual - 1.98)

maelstrom2001, 2017-01-20:
On 170119 can happen everything. Hangups, wrong measurements etc. Bug was found, working on it now.

mittar, 2017-01-20:
OK, thanks, I will re-check with other firmware 6 hours later, when my PC available.

mittar, 2017-01-20:
I am trying 170120 but readings of ohm is bad as on previous firmwares Head 3 (Kanthal A1) - Small Pico original firmware 1.55 ohm - Pico Dual 3.37 170120 firmware ohm - DNA 1.62 ohm.

maelstrom2001, 2017-01-20:
Thank you! Are all firmwares from 170116 to 170120 shows wrong resistance? This issue lasts from beginning of your tests?

mittar, 2017-01-20:
Yes, all of them and from the beginning.

maelstrom2001, 2017-01-20:
Yep, bug was found, thanks a lot! Re-download please [170120](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170120.bin)

mittar, 2017-01-20:
Head 3 - 1.69 ohm ... great , thanks!

maelstrom2001, 2017-01-20:
If you want, fine resistance reading tuning do please with editing Shunt value in Expert settings, if you want to equalize values with DNA. Again, thanks! The main is that resistance not doubled now :)

maelstrom2001, 2017-01-20:
BTW, can you check in Device Monitor output power, output voltage and output current? I suspect that they also incorrectly measured, and the output power may be wrong. Could you please take screenshot of puff like this: [image]

with enabled Power, Power Set, Output Current, Output Voltage and Live Resistance

mittar, 2017-01-20:
[image]

maelstrom2001, 2017-01-20:
Oops... update NToolbox, please: [6.4.34](https://ci.appveyor.com/api/buildjobs/78s1ko4h319i9eb4/artifacts/src%2FNFirmwareEditor-v6.4.34.zip)

mittar, 2017-01-20:
[image]

maelstrom2001, 2017-01-20:
Re-download please 170120 and test again. For safety first puff try to take at half power, 30 W. BTW, 60 W Set on Pico Dual and 60 W on other box with the same atomizer feels equal?

mittar, 2017-01-20:
Yup, feels like same.
[image]

maelstrom2001, 2017-01-20:
Thanks a lot! Now these charts looks more honestly :)

closed 2017-02-03 by TBXin