maelstrom2001/ArcticFox #47: Key repeat issue

joeblowma, 2017-01-14:
When holding a key to repeat, when it steps from first speed to second speed it skips 3 - as far as I can tell in all places (menus, tcr, preheat %). For things like menus rather than values on bigger screen like vtwo the third speed is unusable/useless (not that its even needed.)

The delays between the speed changes seem more than adequate, just occasionally that "skips 3" when going between first and second speed lands at _just_ the wrong time.

Tested on vtwo and pico, though I assume it's core code that would affect all devices - in versions from christmas all the way up to the 170113 nightly.

<crosses fingers hoping the cause is a code bug rather than an interrupt>

TBXin, 2017-01-17:
Can you explain in more details? We can't understand what you suggesting to change in the behavior.

joeblowma, 2017-01-17:
The simplest way to view what I think/hope is just a bug:
-use a big screen device like vtc/vtwo
-press fire and + to enter menu
-hold + (or -) indefinitely to automatically scroll through menus
-observe what happens when autoscroll change speed

it will 'stutter' when it changes from first speed to second and skip 3 menus during the change (this does happen if it is a value and not a menu, but is harder to notice.) When it goes from second speed to third speed, it constantly 'skips' after that (this does not happen if it is a value instead of menus.)

TBXin, 2017-01-17:
I see.

This behavior is almost global for all interactive things. We knew that it's useless in the menu, but reworking this mechanism is a complex task, so we are not touching it.

Behavior about you are talking is a result of the high "navigation" speed and low refresh screen speed. Like "stroboscopic effect".

joeblowma, 2017-01-18:
Ouch, the code to change from first to second speed eats 2 screen refreshes?

Anyway, it is a quality issue and does not affect usability at all - feel free to close as ignore ^^

maelstrom2001, 2017-01-22:
Done in nightly builds.

closed 2017-01-22 by maelstrom2001