maelstrom2001/ArcticFox #48: RX 300 SUPPORT

docdeath, 2017-01-14:
firmware loads  ok still not working as its only seeing one battery in BVO .menu all works fine so
far with 
af_170113.bin

maelstrom2001, 2017-01-14:
Please try this build: [af_170114.bin](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170114.bin)

docdeath, 2017-01-14:
its working now , all as it should  only thing is in nfe tool box only shows 3 battery but apart from that all working sofar

maelstrom2001, 2017-01-14:
Oops :) Thanks, I forgot about it. Please re-download binary.

docdeath, 2017-01-14:
all working as it should do  very good job there

maelstrom2001, 2017-01-14:
Thank you very much for testing and response, keep issue opened until full support will be implemented. Or some new bugs will appear :)

OlafKroll, 2017-01-14:
not work on my RX 2/3 :-)

maelstrom2001, 2017-01-14:
> not work on my RX 2/3

What it means? Can not install firmware, box not start, shows something wrong, or what?

OlafKroll, 2017-01-14:
After install.
Can´t connect via Artifox Configuration.
means... use a newer Firmware

TBXin, 2017-01-14:
For this AF build you need nightly Toolbox build

docdeath, 2017-01-14:
JUST  one thing to ask there is there battery % some where or is not in there . for rx300

OlafKroll, 2017-01-14:
oh sorry my fault

maelstrom2001, 2017-01-14:
@docdeath it should be there selectable:
[image]

docdeath, 2017-01-14:
wops my fault not looking thanks

docdeath, 2017-01-14:
i see you have posted 14a bin any changes for rx300

maelstrom2001, 2017-01-14:
There is a little fix for RX300, maybe not necessary. If it possible, please test too.
14a in main contains bugfixes for Pico Dual.

docdeath, 2017-01-15:
tested 14a  all working as should do  cant see any problems  been using  for 10hrs

Ustes, 2017-01-15:
great work. Tried this on my rx300 with no issues.  Is there any way to show all 4 battery status?

maelstrom2001, 2017-01-15:
@Ustes Thanks! You can see all battery status on Info Screen on 2nd page (+ and - changes page), or switch skin to Circle: Battery Volts line should display voltages consequentially from 1 to 4 while display on.

Ustes, 2017-01-15:
@maelstrom2001 thanks for the info.  Why do you say seeing all 4 battery icons is useless?  I like the 4 cause it shows me if a battery is getting out of balance at a quick glance.  Just my 2 cents. Again, thanks.

docdeath, 2017-01-15:
@Ustes  having RX300 myself  i tested the battery state  and the 4  icons where so out off balance  it was  like maelstrom said worth less and if they are out off balance the mod lets you know in stock firmware and so dose  this firmware so no need for 4 icons  just my 2 cents.

Ustes, 2017-01-15:
@docdeath so what yall are saying is that those 4 icons are not right anyway? if that is the case going back to AF!

docdeath, 2017-01-15:
ustes  in all fairness the icons are there to give you  rough est  off battery life if you want 100%  proof buy dna chip set Wismec don't say any where in info for this mod that battery icons are 100% spot on and never will do   if you feel your being hard done by then feel free to sent mod back  but my self i am keeping mine as i can always use bvo to see battery's

Ustes, 2017-01-15:
@docdeath in all fairness I don't think you understood what I wrote.  I said if what you all are saying is that the 4 icons are never correct, I am going back to flash ArcticFox(AF).  I was agreeing.  Where did I say anything about I want 100% battery accuracy, or that I was being hard done (what ever the fuck that is)?  
Pump ya brakes sun!

docdeath, 2017-01-15:
@Ustes if i have upset you then sorry and feeling hard done by is brit saying for not as you expected

maelstrom2001, 2017-01-16:
@docdeath Hello! Can you confirm that BVO for 4th battery is not working? Thanks in advance!
#53

docdeath, 2017-01-16:
@maelstrom2001  its not working tested

maelstrom2001, 2017-01-16:
Thanks, do you changed BVO with NToolbox or on RX300 in Expert Menu? If you edit BVO in Expert Menu in mod, does it work?

docdeath, 2017-01-16:
i have just tried both  and will not work in both

docdeath, 2017-01-16:
@maelstrom2001  my bad retested in ntoolbox and it works but not working if you use in mod menu

docdeath, 2017-01-16:
@maelstrom2001 that seams to have fixed that BVO  mod menu

maelstrom2001, 2017-01-16:
Nope :) on mod's side did nothing with BVO in today's build. If ain't broke don't fix it :)
Today we should have functional balancer and... charging via USB. Maybe. It's done, but disabled.

docdeath, 2017-01-16:
@maelstrom2001  ok that's strange as if i remove battery's and put new set in it works once then nothing using mods bvo menu

docdeath, 2017-01-17:
@maelstrom2001  when mod is plugged in to usb and charging it do.s  not change screen or shut down to show charging  screen  ,it just stays on main screen and batt icon show its charging  but in really strange movement it go back and fourth then speed up and slow down so have turned off charging

maelstrom2001, 2017-01-17:
@docdeath Thank you for response! It seems that charging constantly switches on and off by itself, and thats why mod never goes to sleep. Will upload some new test version soon.

maelstrom2001, 2017-01-17:
[170117a](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170117a.bin) uploaded, please test it.

Obicom, 2017-01-17:
@maelstrom2001: Any proposal what I can check regarding no charging on my device?

docdeath, 2017-01-17:
@maelstrom2001 the 117a is working but you can not turn off in mod menu or ntool also  it still not going to sleep it gets stuck  as in screen shot 
[image]

maelstrom2001, 2017-01-17:
@Obicom As **docdeath** wrote above, charging routine in 170117 is still very buggy. Something ATM goes wrong, need time to investigate.

@docdeath thank you! Is behaviour of battery icon (charge progress) better than 170117 (without "a") or the same?

docdeath, 2017-01-17:
charge progress is far better with 170117a than 117

Obicom, 2017-01-17:
@maelstrom2001 : af170117a.bin installed and start charging with 670 mA on USB tester.

[image]

maelstrom2001, 2017-01-17:
@Obicom @docdeath thanks a lot, you're great testers, much appreciated!
And 2 more test, please:
- while box charging, switch Stealth Mode on. After dim timeout charging screens should disappear;
- disconnect USB, switch stealth off, switch box off by 5 clicks. Now plug USB in. Box should start charging.

Obicom, 2017-01-17:
1.) works as expected
2.) works as expected

well done

maelstrom2001, 2017-01-17:
@docdeath 
> can not turn off in mod menu or ntool also it still not going to sleep it gets stuck as in screen shot

Well, I think it is right and regular behavior. While mod is charging, it always shows screenshot like you provided. No matter in what state is it - on or off. If it goes to main screen from that on keypress and went back to charging screen after some time - all goes right :) Black screen on USB connect we can get in 2 cases:
- Stealth Mode is on
- batteries are fully charged, so box can go to deep sleep

In other cases charging screen will appear.

docdeath, 2017-01-17:
same here all worked  as above to

docdeath, 2017-01-17:
@maelstrom2001  so can you disable usb charge from mod menu and ntool box still because  i cant  in ntool i untick box  and upload and still charging  unplug usb replug in still charging  open mod menu with out usb plugged in  usb charge is off , but still charging  when i plug usb in

maelstrom2001, 2017-01-17:
@docdeath Yes, this option will be implemented for sure. Need a little break for refactoring code, and then I'll ask you to test build with it. Maybe even tomorrow :) for now you can use previous builds with disabled charging in hard code.
In 170117a build indication of charging is working, but physically box should not take current from USB when "USB Charge" option is disabled.

maelstrom2001, 2017-01-18:
[170118](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170118.bin)
Refactored, please test. "USB Charge" option also should work.

docdeath, 2017-01-18:
yes that works ok can turn  on and off in ntool and mod menu

Obicom, 2017-01-18:
and with USB current tester it works fine ... charge off = 0mA, Charge on = 680 mA

maelstrom2001, 2017-01-18:
Thanks a lot for responses! What you think, we have to implement stock Wismec charging algo up to 1,5 amps or staying on "old" 1,0 amp limit?

Obicom, 2017-01-18:
With four batteries I would prefer the 1,5A current to get faster charging results.

Obicom, 2017-01-18:
@maelstrom2001 : Is Avatar charging adapter support in focus?

docdeath, 2017-01-18:
@Obicom  really? you charge in mod

docdeath, 2017-01-18:
@maelstrom2001  to be safe i would stick to 1.amp

Obicom, 2017-01-18:
@docdeath : Normally not but if I would do I would prefer 1.5A ;)

maelstrom2001, 2017-01-18:
> Is Avatar charging adapter support in focus?

In near future - no. We have a little problem with ROM size, we should not exceed 102400 bytes limit, and don't know how many space will be occupied by Avatar routines. Maybe later firmwares will be separated for large and small screens, or something else... Don't know. So, it will not be implemented in future release or two.

@docdeath By the way, is 170118 crashing you PC on connect?

docdeath, 2017-01-18:
nope none off  them have caused problem  here

maelstrom2001, 2017-01-18:
Charge current is restored to stock in 2 variants:
[170118a](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170118a.bin) - 1 amp limit
[170118b](https://dl.dropboxusercontent.com/u/43399149/VTC/arcticfox/nightly/af_170118b.bin) - 1,5 amps limit

1,5 amps may be not reached depending on USB source and batteries voltage.
Please test with USB tester, if possible. Thank you!

Obicom, 2017-01-18:
Currently no more crashes with af170118
Test af170118b: 87% charging level and charging current = 1440 mA

Ustes, 2017-01-18:
@maelstrom2001 Great job!!

[34 more replies]

closed 2017-02-03 by TBXin