maelstrom2001/ArcticFox #50: iStick Pico 75W Brick

Obicom, 2017-01-15:
Dear Creator,

I tried to update a iStick Pico 75W with Toolbox 1.4 an af170103. After update completed display still black. Reconect the Pico to NFE TB device is show as Wismec Reuleaux RX 2/3. Tried oder af161224.bin but same effect. Eleaf updater do not recover the Pico as a Pico so I am not able to downgrade. Now my Pico is bricked. Any recommendation  what I could do or try in this case?

Best regards, Dirk

Obicom, 2017-01-15:
Ok, I was able to downgrade to original FW 1.03 from Eleaf via the NFirmwareEditor -> Tools -> Frimware Upgrade ... afterwards I ried again to install af170103.bin and the same issue happens. Display black and Toolbox shows a Wismec Reuleaux 2/4 device.

maelstrom2001, 2017-01-15:
Close NToolbox, remove battery from mod, hold + button on mod, connect to PC and flash stock firmware with stock updater. Then we'll continue :)

Obicom, 2017-01-15:
Is a reported Pico is back to life but I am not able to install AF to a Pico .. could you confirm this issue?

maelstrom2001, 2017-01-15:
> could you confirm this issue?

No, because I have Pico HW1.00 and never faced problems while updating. Re-download NToolbox 1.4 and af170103.bin from our repositories, make "clean install" (run it from new folder), and try again.
On my side, I'll do the same right now.

TBXin, 2017-01-15:
Assumption: may be you are connected and pico and rx2/3 to one pc, simultaneously? :)

dywersant, 2017-01-15:
Maybe OP has different HW rev version that hasnt been tested before ?

TBXin, 2017-01-15:
No matter what HW version is, neither firmware nor NFE can't rewrite product id (pico, rx2/3 and so on..)

Obicom, 2017-01-15:
See the details in a short video on my channel (not public). Same issue with fresh download and installation. BTW on the same channel you could find my review about NFE TB and AF

https://youtu.be/7Z-lsNFQ-ug

TBXin, 2017-01-15:
@Obicom how much joyetech/wismec/eleaf devices connected to your PC simultaneously?

Obicom, 2017-01-15:
only one at one time ... but did updates on different devices before

Obicom, 2017-01-15:
Now a new problem ... the NFirmware Editor recover teh Pico as a RX 2/3 (v 6.3 and 6.4) and do not want to downgrade my Pico. If i put out the batterie and press + the same issue. Also with Eleaf software. Currently no chance to come back to original FW

Obicom, 2017-01-15:
@maelstrom2001: The tip "Close NToolbox, remove battery from mod, hold + button on mod, connect to PC and flash stock firmware with stock updater. Then we'll continue :)" do not work ... any other idea how I could rescue my Pico now?

TBXin, 2017-01-15:
@Obicom open Firmware Updater, press and hold Ctrl+Alt buttons, then press Update from file and select pico original firmware file, when updating process will start you can release ctrl+alt buttons.

This key combinations will disable firmware and device compatibility checks.

Obicom, 2017-01-15:
Thank you for the advice .... this works ... I tried it now qith a different Pico on a other PC ... exactly the same is happening. NFE must change HW id ... Wismec software also recovers the Pico as a Releuaux RX 2/4 and I was able to install with the Wismec updater the stock R2/3 firmware.

TBXin, 2017-01-15:
@Obicom definitely this is a very strange situation.
But i am sure that NFE will not modify HW/PID until user will not ask about it. NFE products more than 1 year "at market" and you are the first, who is facing this kind of issue.

We are trying to understand what is going with your pico's and what can cause this...

Obicom, 2017-01-15:
Therefore I am a perfect beta tester ... ;) I do comparable work at work and I am a "lucky guy" who find error nobody else find ;) if you want to investigate more of this issue please let me know how I can help to find the problem.

maelstrom2001, 2017-01-15:
> Therefore I am a perfect beta tester ... ;)

True!!! :-D

OK, now seriously. Download attached file, unpack and try to flash with NFE with disabled compatibility check if box detected as Wismec, or with Eleaf flasher, if box think that he's Pico.
[af_170103e.zip](https://github.com/maelstrom2001/ArcticFox/files/706845/af_170103e.zip)

Obicom, 2017-01-15:
I am not sure if I 100% understand what you want. Is was able to install the stock firmware with the tip from TBXin (Ctrl+Alt) .... if I install your version af_170103e.bin same issue with Wismec RX2/3 but I was able to recover to stock firmware with NFE. I am confused why this happend on to of my systems (Windows 10 64 bit) with two Pico mods but you are not able to reproduce the issue.

Obicom, 2017-01-15:
Tried know on a third PC ... same behavior .....

maelstrom2001, 2017-01-15:
Install stock firmware to bring the box up. Pico should be detected by Eleaf flasher as right device, and it should work. And now install with **Eleaf flasher** (not NFE) af_170103e.bin.

Obicom, 2017-01-15:
ok, did it .... first flashed stock firmware ... Eleaf flasher detected Pico as a right device with FW v1.03 .. then flashed with Eleaf flasher the af170103e.bin .... afterwards Eleaf flasher detect a unknown device ... NFE detects a Wismec Reuleaux RX2/3 ... same issue  :(

maelstrom2001, 2017-01-15:
Well... Please start NFE Updater (no matter, NToolbox or NFirmwareEditor), remove battery from Pico, hold + and connect it to PC with flasher running. And take a screenshot of it. We need to know what device will be detected in LDROM Boot Mode.

Obicom, 2017-01-15:
BTW if you need an additional beta tester fpr RX300 device let me know ... 
As you can see in my RX300 review I was not really happy with TC control on new RX300

Obicom, 2017-01-15:
First of all it doesn't work with + key (only eleaf logo is flashing) but with - key
Then it is detected as a Wismec RX 2/3
With batterie inside as a istick 
[image]
Pico

maelstrom2001, 2017-01-15:
> it doesn't work with + key

And if hold - (instead of +) and do the same?

Obicom, 2017-01-15:
as I wrote above .... then it is detected as a Wismec RX 2/3 (see screenshot)

maelstrom2001, 2017-01-15:
> With batterie inside as a istick

Without battery, please

TBXin, 2017-01-15:
@Obicom did you upload original eleaf firmware before doing this?

We want to know how your bootloader is working.
You need to upload original FW and get mod working.
After do all steps again, remove battery, hold plus and only then connect to the USB.

Boot mode should be LDROM, not APROM

Obicom, 2017-01-15:
Yes of course ... without batterie .... hold - key and NTF TB flasher detects a RX 2/3

I did exactly what you said ....
First flashed original v1.03 FW
Put batterie out , press +, eleaf logo is flashing and NFE TB flasher is detects in rythem (on/off)
Put batterie out , press -, NFE TB flasher detects a Wismec RX 2/3 but shows APROM bootloader not LDROM mode ...

TBXin, 2017-01-15:
@Obicom ooookay, i see.

Did you bought your picos or eleaf sent it to you for review?

Obicom, 2017-01-15:
I bought the most of them (have four) ... I am not sure if one directly from Eleaf ... I tried know two devices ... shall I do with more?

maelstrom2001, 2017-01-15:
And the last question: is that Picos are pre-production samples?
'Cause their LDROM's are too different from mass production devices.

Obicom, 2017-01-15:
Ok tried now a third one .... and this device works correct .... ArticFox FW update succesfull
Maybe I have selected two early devices ... is it possible to fix this? If not - not a big deal

TBXin, 2017-01-15:
@Obicom eleaf/wismec/joyetech or someone... who is creating PCB's made a fatal mistake when creating this boards. They are flashed bootloader from RX2/3 to the pico's boards. Thats why ArcticFox is think that he is running on RX2/3.

Theoretically this almost hardware issue can be fixed, but it's dangerous operation.

maelstrom2001, 2017-01-15:
> is it possible to fix this?

Restore to stock and use it. AF is need to be modified to work on these boxes, and if you notice, "nightly" versions appears daily, 'cause development flows. It will require to do special builds for you only.

Obicom, 2017-01-15:
Where could I find the nightly AF builds and nightly Toolbox for testing? (special for RX300)

maelstrom2001, 2017-01-15:
[Welcome!](https://www.dropbox.com/sh/vuzq4vk4ypld818/AACtsDNzxa2zPXhT-RhnrPZia?dl=0)
And many thanks for published review :)

Obicom, 2017-01-15:
Thank you ... to configure nightly build a special nightly NFE Toolbox is needed or works with the current version?

maelstrom2001, 2017-01-15:
Nightly builds are compatible with nightly NFE Tools only. And vice versa. Release builds - with release versions of NToolbox.

Obicom, 2017-01-15:
and where could I find the nightly builds of NFE Toolbox please?

maelstrom2001, 2017-01-15:
[AppVeyor](https://ci.appveyor.com/project/TBXin/NFirmwareEditor/branch/master/artifacts)

Obicom, 2017-01-15:
Thank you ... works well on RX300 :)

closed 2017-01-16 by maelstrom2001