maelstrom2001/ArcticFox #51: [Suggestion] Add custom TFR materials or allow renaming of existing materials

Sanaki, 2017-01-15:
When doing complex builds with multiple wire types involved or using exotic types not listed, there can be no proper association available, leaving the requirement to repurpose something with an unrelated name. While this isn't necessary by any means, it would be nice to be able to have a couple custom materials (even if you can't rename those materials beyond CSTM1/CSTM2, etc). Being able to rename existing materials would accomplish the same goal for all intents and purposes.

TBXin, 2017-01-15:
Hmmm...

You can rename all 8 TFR materials, also you can export curves and import them, even more - you can download curve from steam-engine and import to your box.

[image]

[image]

Sanaki, 2017-01-15:
Well crap, I didn't see the profile rename option in there. I've been using steam engine imports for my coils as it is. Not sure how I missed that. Thanks. Closing this one, I'm more than happy with that.

closed 2017-01-15 by Sanaki

Luminescah, 2017-03-30:
Sorry for reviving, but I'm having a problem when I try importing custom .csv from steam-engine.
http://imgur.com/CBVUrHQ

maelstrom2001, 2017-03-30:
@Luminescah steam-engine has changed export format, please update NToolbox to [the recent nightly](https://ci.appveyor.com/project/TBXin/NFirmwareEditor/branch/master/artifacts)

Luminescah, 2017-03-30:
Works fine. 
Thanks!
One quick question, I don't think this warrants an extra thread. 
On the homescreen, how can I quickly change the temperature? Plus/minus seems to only change the wattage.

maelstrom2001, 2017-03-30:
Set Temperature Dominant option in NToolbox for current profile:

[image]

or assign Multiclick to toggle this option:

[image]

To change non-dominant regulation press Fire 4 times - you'll enter in edit main screen mode, it's the same as 3 clicks in stock Joyetech firmwares.

Luminescah, 2017-03-30:
Thanks again.
Great firmware, much appreciated for your work.