maelstrom2001/ArcticFox #53: BVO for with RX300

Obicom, 2017-01-16:


maelstrom2001, 2017-01-16:
Strange... Is battery voltage B4 changing on BVO screen while you changing its offset?

Obicom, 2017-01-16:
I hope understand your question right. If I change the values for all four batteries on BVO with Toolbox yes the offset value is changing on screen.  Afterwards I checked my battery values on the "Info screen on the device" and B1-B3 are changing but not B4.

TBXin, 2017-01-16:
@Obicom thanks for feedback. Thats NFE Toolbox issue.

TBXin, 2017-01-16:
@Obicom try, please, [this build[(https://ci.appveyor.com/project/TBXin/nfirmwareeditor/build/6.4.20/artifacts).

Obicom, 2017-01-16:
@TBXin : With the new nightly build of TB (6.4.20) the issue is solved. Issue can be closed. Thank you :)

TBXin, 2017-01-16:
Nice!

closed 2017-01-16 by TBXin