maelstrom2001/ArcticFox #59: [Vtwo Mini] Changing batteries, resetting stats and confirming "old coil" causes firing at 30W

forkoz, 2017-01-16:
I'm running the 1/16 release on my vtwo mini and using a 1.4 ohm coil with SS316L. When I change batteries, reset stats and keep the old coil the mod fires at 30W instead of the 10 it was set to. I have to go into toolbox and re-upload the profile for it to stop. If I select "new" coil it gives a TCR error and will not fire.

from: https://github.com/TBXin/NFirmwareEditor/issues/130

It keeps firing at 30W while I hold down the button, huge vapor and burnt smell. Resistance at rest is 1.43ohms  I disabled TCR check so you are right that it comes from there.

maelstrom2001, 2017-01-16:
> disabled TCR check

Did I understand you correctly - the problem was solved?

forkoz, 2017-01-16:
"solved" as in I turn off TCR check. Otherwise it keeps firing at 30W even if you lower the temperature. I will try a 1.1 ohm next.

forkoz, 2017-01-16:
At 1.04 ohms the device fires very quickly at 30w and then stops. So if you build a coil technically <1.5ohms but its hot or the resistance changes the TCR check will fire indefinitely and cook things.

TBXin, 2017-01-17:
This is a default behavior for any firmware. As maelstrom2001 says - you can disable TCR Check and device will never try to check coil with a 30 watts warmup.

forkoz, 2017-01-17:
IMO, it should still fail or stop after 1s to match the 1 ohm behavior. There must be something goofed for it to just keep firing. What's worse is that the logger doesn't appear to record that the 30w firing is happening.

maelstrom2001, 2017-01-17:
> it should still fail or stop after 1s

It is not acceptable for complex builds. Too few time to warm up, for example, fused claptons, and it constantly will rise TCR Errors. And I didn't mention any strange things like notch coils...
Summing up: you know what coil you install to atomizer, but box can only guess what exactly it is.

forkoz, 2017-01-19:
I don't know what it is checking exactly. Resistance rise? I set TFR and material manually and don't have smart mode on. There is only VW profile and this one for SS316L

TBXin, 2017-01-19:
You can't select any material while your profile in the VW mode, so, your porifle in the Temp. Control mode and thats why box is checking resistanse rise.

forkoz, 2017-01-19:
It did it even when puling the battery in the temp profile. Are you saying if I disable VW, leaving one profile, it will stop?

TBXin, 2017-01-19:
em... Battery temp profile... WAT?

[image]

That's all you should do to get rid of 30 watts starting boost.

forkoz, 2017-01-19:
you misunderstand, this happens when I pull the battery out resetting the device while the current profile is TC. I was asking if the TCR check was caused by having a VW and TC profile registered in the device.

TBXin, 2017-01-19:
TCR check appropriate only for TC mode. If your active profile in the VW mode - you will never get TCR checks and TCR errors.

maelstrom2001, 2017-01-19:
> 1/16 release

What is it? Update to [170103](https://github.com/maelstrom2001/ArcticFox/releases/tag/170103)

forkoz, 2017-01-19:
I'm at 170116. Forgot that you write D/M/Y and I write M/D/Y. 1/16 is January 16th. If I only use VW of course I get no TCR check, same if I turn it off. Not saying there is no work around.

But TCR check should not continuously fire 30w towards the top end of the resistance range especially when everything is set up in the profile and on top of that write 10w in the device monitor. Not everyone is sub ohming at the highest power possible.

maelstrom2001, 2017-01-19:
> same if I turn it off

I. e., when you turn Check TCR off everything is OK, and box will start from right power value?

forkoz, 2017-01-19:
basically. I turned off TCR check completely so I wouldn't get a 30W surprise vaping in TC mode and changing batteries.

closed 2017-01-19 by maelstrom2001