maelstrom2001/ArcticFox #62: Egrip 2 LED light support

AAI688, 2017-01-17:
Sorry if i bombard u with questions and features. I got an egrip 2 coming and i was wondering if the led light support would be available as i dont see it in the menu structure. Would it be easy to add? Cheers for it.. ^_^

dante-35, 2017-01-24:
Evic Aio has same feature too.
Its very handy to see liqud status in dark places.

StillVoidingWarranties, 2017-03-06:
I own a Joyetech eGrip II Light and would love to see support for the tank light. I volunteer to test any beta releases of ArcticFox.

maelstrom2001, 2017-04-14:
Will be added to upcoming release as well as for eVic AIO and Pico RDTA.

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001

AAI688, 2017-05-09:
hi just tested the new nightly 09. where is the LED light settings? i cant seem to find it??

maelstrom2001, 2017-05-09:
Hello! There is no settings for this option, LED should light while you press Fire button.

komaramnt, 2017-05-10:
I opened new tread, but I will ask in here... can we get a option ot turn it off or change colors please........ I know it is hard to do things for everyone but if we can help with something just name it

sorry for my bad English