maelstrom2001/ArcticFox #63: Current nightly build crashs my PC

Obicom, 2017-01-17:
Every time if I connect my RX300 with AF nightly build af170116 or af170117 together with TB 6.4.20 my PC do a reboot. If the box is still connected during the reboot and I start the TB after Windowes starts it works well. Disconnecting - connecting again ... same issue .. PC reboot (also without started TB 6.4.20)

maelstrom2001, 2017-01-17:
If you connect RX300 to PC with **no** NToolbox running, is everything OK?
Ask that because we started charging over USB support in nightlies, and reboots can be caused by overloaded USB Port. If PC crashes when NToolbox running and you connect any (maybe, single battery mod) with last nightly, it's definitly software problem.

Obicom, 2017-01-17:
It happens also without TB running and with 1f170116 (without charging function)

maelstrom2001, 2017-01-17:
Is 170116 on another box, for example, VTC Mini, also crashes PC?

Obicom, 2017-01-17:
No, it happens only with RX300.
With VTC Mini works with af170116 and af170117 without crashing PC.

maelstrom2001, 2017-01-17:
Can you measure RX300 (builds 170114, 170116, 170117, maybe stock Wismec firmware too) power consumption on USB? With USB tester, and from any standalone USB charger.

docdeath, 2017-01-17:
i have not had any problems with my pc and have used all firmware  and when usb plugged in it draws  500ma and on stand alone charger  it will draw 1amp in wismec firmware  on AF its 500ma on both

Obicom, 2017-01-17:
Now I am confused .. currently no crash with af170117.bin (before every time) BUT USB current tester shows zero mA (with other mods shows correct current) but on RX300 display battery symbol flashing with flash icon inside ....

Obicom, 2017-01-17:
@docdeath: did you checked with a RX300? I get no charge current with af170117

Obicom, 2017-01-17:
@maelstrom2001 : next test, next reboot :( but not charging current (Bat settings on LGHG2 profile) ... changed to generic profile ... same .. no charging current

maelstrom2001, 2017-01-17:
@Obicom install Wismec firmware and check current on it, maybe even with other batteries.

Obicom, 2017-01-17:
Original stock FW 5.04: USB current tester shows 480mA on my PC USB slot

maelstrom2001, 2017-01-17:
I offer to continue messaging in #48. Let's keep all development issues in one RX300 thread, OK? :) I think it will be more organized and productive, isn't it? Also, the support is still under development, and it doesn't apply to the released versions.

maelstrom2001, 2017-01-18:
Seems to be fixed, so closing. Thanks a lot for testing :)

closed 2017-01-18 by maelstrom2001