maelstrom2001/ArcticFox #70: VTwo battery sudden drop

AAI688, 2017-01-20:
Hi All, Vtwo battery suddenly drop on nightly 18b. Was around 50% auddenly drop to 10% and showing weak battery. Battery profile is Gen. What profile should it be? Or is this a bug?

maelstrom2001, 2017-01-20:
What battery exactly you  are using? How old it is? What output power value is?

AAI688, 2017-01-20:
Battery is inbuilt. 5000mah i believe.. 3 days old.. lol.. but that shouldnt matter right? Where can i see the output value?

maelstrom2001, 2017-01-20:
Battery profile should be set to GEN. Try another nightly or release build, need more information to understand when problem appear. Thank you.

AAI688, 2017-01-20:
What information u need? Other mods i got which is internal batt (around 2000mah) are running okay on ur firmware.. only this one thats suddenly acting wierd.. i might try other nightly and see if its any change..

maelstrom2001, 2017-01-20:
> What information u need?

Ideally need 2 neighbor builds numbers, for example, 1701xx works OK, but next 1701yy has problem.

AAI688, 2017-01-21:
so previous 170108B has problem. currently im testing 170109.. half day in and its still 100% has not dropped yet.. vaped probably about 20-30 min worth by now.. battery profile Gen still.. im sure when using other mod it would hav dropped to 98%-88% and so on.. strange how its measuring the battery.. i will keep updating on the battery...

maelstrom2001, 2017-01-21:
Oh... Got it. Completely forgot about VTwo. GEN profile should use Li-Po profile, like Cuboid Mini, eGrip II etc. You're right, ATM we have wrong discharge curve.
Today I'll fix it, look in 3..4 hours for nightly build. Thank you!

AAI688, 2017-01-21:
end of the day and i got 71% which i think its should be around 50% as with a 5000mah/2500mah X 2 (25r) should last me 2 days worth of vaping for my usage.. anyway i just flashed ur new nightly 170121 and straight away the battery falls to 61% which should be roughly correct.. i will monitor that tomorrow and hope it is fixed.. cheers for the quick nightly...

AAI688, 2017-01-22:
finally managed to drain the battery on the vtwo in around 1.5 days.. but anyway the percentage seems to be reading quite correctly now as it dont drop suddenly as per previous nightly build.. this build 170121 is okay now using the Gen profile.. i am charging it now and will try another discharge.. at least i think its on the right track. thanks..

closed 2017-01-23 by TBXin