maelstrom2001/ArcticFox #71: Lock Fire Button

heru21, 2017-01-20:
hey admin,

i hope for next update frimware include features **lock fire button** 

for example if device in standby mode for 20sec device made auto lock fire button
i think it be more safety features if we forget to turn off device and we put device to pocket or vapor bag 

thanks

TBXin, 2017-01-20:
And how this feature should work?

Okay, after 20 sec delay fire button will be locked - how user should unlock it?
E.g. user should press fire button a few times? there is already working solution - 5 clicks to turn on/off device.

heru21, 2017-01-21:
**active in click menu 2x click is an empty slot it can be add in 2x click to lock fire button on/off and still be customize** 

i think add this feature in click menu add some lock fire button its a great deal its like a DNA if have a lock fire button...

**_this firmware its can be kill DNA and YIHI_**

_i use for a week, i think use this frimware its a great choice favor production ok, its great read temperatur and most i like Preheat setting its so so great_

i use **Evic Mini Vtc** and **Wismec RX200s**
[image]

heru21, 2017-01-26:
2 click to on/off lock fire button and still be change if user dont want it
from control to change that and 5 click to on/off device

madotha, 2017-02-01:
I'd also welcome this, that you can edit a 2,3,4 click to lock the fire button and adjustment buttons.
Just like on a YiHi chip, where you can 3-click-lock device (although there's the shutdown function hidden in the 5-click-menu)

heru21, 2017-02-23:
Yes but it more like dna chip... When we have a lock fire button for more
safety...

I'd also welcome this, that you can edit a 2,3,4 click to lock the fire
button and adjustment buttons.
Just like on a YiHi chip, where you can 3-click-lock device (although
there's the shutdown function hidden in the 5-click-menu)

tiryaki92, 2017-04-10:
It is not in the click menu. As with some Smok devices, the + and - keys and the fire button must be locked when the + and - keys are pressed for a quarter of a second. This is much more useful

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001