maelstrom2001/ArcticFox #72: Wrong power value after config update with nightly-firmware

hawkeyexp, 2017-01-20:
Situation:
3 profiles (2 ni200 / 1 ss316)
Active profile is number 1
After adjusting used power in profile 1 via software it is always wrong set to 43,8W instead of 50w which is set in profile.
power values of profile 2-3 (inactive) are written correctly. Tried on cuboid and evic vtc mini - same behaviour.
If active profile is 2 or 3 the error happens always on the active profile when trying to write new power value.

TBXin, 2017-01-21:
Can't reproduce.
What's your coil resistance, which you are using with profile 1?

hawkeyexp, 2017-01-21:
0,12 ohm

TBXin, 2017-01-21:
Can you reproduce this on the latest builds of AF and Toolbox?

hawkeyexp, 2017-01-21:
Seems to be still present .... Here is a screenshot and my config...
[https://storage.hawknetwork.de:8443/index.php/s/6rqoczo7pCAKJs7](url)

TBXin, 2017-01-21:
Screenshot is nice 😄 
Your device?

hawkeyexp, 2017-01-21:
yes :-) sorry - was to late - should show values - update is online

maelstrom2001, 2017-01-21:
Can explain why you got 43.8 watts, other - don't know.
You locked 0.08 ohms in profile. Every time you upload settings, box re-read atomizer resistance. If it was read 0.07 ohms, it will set output power according to limits, in your case it is output current.
[image]

No matter what resistance you locked and edited in profile - atomizer will be checked anyway, and limits will be set according to real, not saved/locked resistance.

hawkeyexp, 2017-01-21:
Not thinked about it :-) But does it make sense ? On device i can set it without problems to 45 watts which will be acceptet - behaviour should be changed - possible a hint in software that watts could not be reached caused by current limitation etc. or it should do same on device etc. - i think somebody will be wondering like me why it doesn't accept setting in profile :-)

maelstrom2001, 2017-01-21:
Yep, such behavior is not obvious, but... You can unscrew atomizer, set any power you want, and then screw atomizer on. What will happen? Box should decrease power by itself to its possibilities. About the same thing happens when uploading profiles.

hawkeyexp, 2017-01-21:
you are right - but i think you know it can cause confisions :-)

maelstrom2001, 2017-01-22:
No problem, will explain again and again. There' s not much we can do :)

closed 2017-01-22 by maelstrom2001