maelstrom2001/ArcticFox #73: iStick Pico: no screen, no function after flashing

MartinEmrich, 2017-01-21:
Just flashed my iStick Pico with Build 170103. After that, it reports "Wismec Reuleaux RX2/3" as device name, and neither screen nor buttons work anymore.

I cannot flash back to the original firmware: the eLeaf Updater reports "unknown device", and the NFEToolbox Updater reports "Opened firmware is not suitable for the connected device".

Can I "force" the original eLeaf firmware on the device somehow?

TBXin, 2017-01-21:
Hello,

Bad news for your. Your Pico is defective, manufacture issue.
You can read same problem issue: https://github.com/maelstrom2001/ArcticFox/issues/50

You can install original firmware with "force". Before pressing "Update from file" button press and hold ctrl+alt keys, choose firmware file and when update process is started you can release keys and wait.

MartinEmrich, 2017-01-21:
Awesome, that did work. Now I am back to stock iStick firmware. I read #50, looks like the same issue. (Fun fact: @Obicom did the awesome german Youtube review that brought ArcticFox to my attention in the first place :)

closed 2017-01-21 by maelstrom2001