maelstrom2001/ArcticFox #79: Pico Mega limited to 80w

Ersad26, 2017-01-22:
Hello,

I was using my Pico mega with 100w  on the original v 1.01 firmware
I just installed the 170103 and power is limited to 80w
Is there a way to pass 80w limit ?

Thanks

maelstrom2001, 2017-01-22:
The recent [Eleaf firmware for Pico Mega](http://www.eleafworld.com/softwares-for-istick-pico-mega/) has 80 Watts power limit. Are you vaping at 100 W? And what is battery life?

Ersad26, 2017-01-22:
Battery life is almost same with vtc5 batteries

You should try [this](http://www.eleafworld.com/wp-content/uploads/2016/05/iStickPico_Mega_V1.01.zip) version for 100w output

maelstrom2001, 2017-01-22:
I don't know why Eleaf reduced power limit in recent firmwares. Maybe just because electronics can't give 100 W, maybe for some safety reasons or something else. But they did it. Only v 1.01 has 100 W limit, 1.02 and 1.03 has 80 W. I am inclined to believe that 100 Watts was a mistake.

Ersad26, 2017-01-22:
That means no 100w for me than 😢

I believe they released a 100w version just for marketing purposes. Some people say 80w version is using the same board with the 100w version.

maelstrom2001, 2017-01-22:
Asking again: what is your daily vaping power? Are you really vaping at 100 W on your Pico Mega?

Ersad26, 2017-01-22:
Yes,i use 95-100 watts daily with that device

maelstrom2001, 2017-01-22:
We can set power limit up to 500 Watts, and box will display anything what you want. But 100 W from 1 cell sounds unreal, just beleive :) Chinese watts and real watts are slightly different. Even boxes with LiPo batteries are limited to 80 Watts, and LiPo cells has output current higher than any IMR.
Box can show you 95..100 Watts output, but not the fact that it **really** give it.

Ersad26, 2017-01-22:
I only use this mod for fun actually.
I was torturing it for a long time with water, mud, drops from 1meters and i did skydive from 2400 meters with mod and melo 3 tank in my pocket. the thing was covered with juice.

I'm planning to do a custom battery holder with 2 batteries in parallel.

I'll be glad if you help me with torturing that little monster with more power. 😄

BigBadBlackWolf, 2017-01-22:
@Ersad26
i hope you know what you want doing! boom

2 batterie cells in parallel are not for more power, only for longer life!

Ersad26, 2017-01-22:
> 2 batterie cells in parallel are not for more power, only for longer life!

I know 😄 i have a holder with 2 battery in series. the mod already let the magic smoke out for like 4 times. 

I have a special power supply that can output 4.4 volts and 32 amps at the place i work.
Normally we use it for testing relays.

BigBadBlackWolf, 2017-01-22:
@Ersad26
if you charge the akkus, don't charge this over 4.2V! Cells can't destroyed!
Cells have 4.2V only for a shirt time after charging. 3.7V are average for
cells.

BigBadBlackWolf, 2017-01-22:
@Ersad26
sorry, cells can destroyed!

Ersad26, 2017-01-22:
I just use the power supply without batteries when using it.

I know the limits of cells. I know the risks of overvoltage, overcharging and i know what a battery can do when abused. The environment is safe. I use this mod for testing purposes.

It would be better if we get a wattage limit of 100 or 120 but it's not that important. the device can only support 90 watts for 10 seconds without burning the board. 😢 

Thanks for support.

BigBadBlackWolf, 2017-01-22:
@Ersad26
better you buy a new e-cig with 2 cells in serie and 1 or 2 cells parallel
to, for more power and longer life!

closed 2017-01-22 by Ersad26