maelstrom2001/ArcticFox #80: required lower temperature eleaf pico

golijeevankumar, 2017-01-22:
can i get lower temperature than 100 c that will help me 
starts from 20 c 
mod eleaf pico 75w

TBXin, 2017-01-22:
Ehm... in your room temperature is higher than 20c.

maelstrom2001, 2017-01-22:
> starts from 20 c

For what purposes?

golijeevankumar, 2017-01-22:
I can't vape from 100 c it's too high for me and I want to vape in low
temperature

TBXin, 2017-01-22:
Did you know, that water has 100c boiling point? >__<
Vape liquid have a higher boiling point, so, how will you vape without steam?

golijeevankumar, 2017-01-22:
I can't take 100 c temperature so asking if I can lower than 100

TBXin, 2017-01-22:
Okay, you can.

Step 1: Open your liquid bottle;
Step 2: Start breathing;
Step 3: ???
Step 4: Profit!

golijeevankumar, 2017-01-22:
Thanks for your suggestions
If u can't help me out don't give me all nonsense methods

closed 2017-01-22 by TBXin

sandbagjohnny, 2017-01-22:
He's trying to tell you that your request doesn't make sense because ejuice barely vaporizes at 100c to begin with. I've never seen any production device (including the DNA boxes) myself; There's a reason for that :)

Hunterror77, 2017-02-06:
resistance is not lock correctly.... this is the problem. You do not need to go any lower than 100 c.