maelstrom2001/ArcticFox #81: TC Mode temperature too low on RX200S

RicKro, 2017-01-24:
I am using the TC Mode on a iStick TC100W and on a RX200S. Referring to the original Firmware the TC Mode is not working quite good on the RX200S. I need to put the temp much higher then with original Firmware or on the TC100W to receive the same amount of vape. Usually I vape in TC with 220°. If I put 220° on the Arctic Fox Firmware on my RX200S, there is almost no Vape. When I do the same on the TC100W it is similar to what both mods are doing with original Firmware. I use only SS316L wire. I also tried with TFR and TCR but no change.

maelstrom2001, 2017-01-24:
Compare atomizer resistance reading on iStick and RX200S. Due to schematics differences boxes can detect and save "cold resistance" (which is base value for calculating atomizer temperature) differently.
For example, I have to set Shunt correction on VTC Mini to 94%, while on Pico that value is set to 102%. When atomizer will detected on both boxes with the same resistance, temperature settings will be the same also.
If you find that TC works incorrectly, first you should check "cold resistance" saving. In your case "cold resistance" is lower than it need to be, and atomizer temperature will be calculated with overstatement.
Also note that different boxes has different connectors, and it also (sic!) can lead to slightly different atomizer reading while you screwing atty on the box.

RicKro, 2017-01-24:
the resistance is detected correctly. I also know of course that the cold resistance is important. I assume I did everything correct, but still the TC is not working correct on the RX200S. I used the same atomizer on both boxes,

maelstrom2001, 2017-01-24:
Run Device Monitor on both boxes with the same TC settings (material, preheat etc.) and pay attention on Live Resistance while firing. Set Y scale to 50% for example and compare resistance rising dynamic on iStick and RX200S. Maybe RX200S resistance measuring is higher than it should to be.

TBXin, 2017-01-24:
@RicKro try to use custom TCR: 120 with your wire.

OFW SS316 Mode - is a TCR 120, this value is totally average for all kinds of SS.
ArcticFox SS316 - is TCR 92, real SS316 TCR.
Even more, we have TFR SS316L which is much better than linear TCR.

RicKro, 2017-01-24:
I made a test with the same atomizer on both boxes with waiting to cool down between the tests. the output of vape is MUCH more on the TC100W then on the RX200S. attached the recorded device monitor during the test. I have set the TCR Mode with a TCR of 94, which should be exactly what I need for my wire.

[TC100WRX200S_TC_Test.zip](https://github.com/maelstrom2001/ArcticFox/files/726314/TC100WRX200S_TC_Test.zip)

TBXin, 2017-01-24:
Look at the resistance on both devices.

RX200S reads a higher resistance, than iStick - it directly affects the calculation of the temperature.

RX200S:
Res 0.366, Temp 240, Temp Set 220 - and power are going down to limit temperature.

iStick
Res 0.353, Temp 190, Temp Set 220 - temp is not growth to your limit so device will increase power to your set (70w).

All answers in your csv files.

RicKro, 2017-01-24:
but how to manage this to work?

maelstrom2001, 2017-01-24:
In addition: screw on cold atomizer on iStick and simply start Device Monitor and watch Live Resistance. Don't fire, just look at charts. And then change the box and do the same. What Live Resistance will be shown in standby on iStick and RX200S? Locked/saved resistances are pretty close, but what about Live?

TBXin, 2017-01-24:
Menu -> Settings -> Advanced -> SHUNT, set it to 97%

RicKro, 2017-01-24:
The Live Resistance on both boxes with cold atomizer is exactly the same.

[TC100WRX200S_TC_Cold_Resistance.zip](https://github.com/maelstrom2001/ArcticFox/files/726378/TC100WRX200S_TC_Cold_Resistance.zip)

maelstrom2001, 2017-01-24:
So there is only one explanation: 200S read resistance higher than TC100 while firing. Look at resistance rise on both boxes. On TC100 it is much smoother, while 200S very quickly goes into protection.
Use Shunt Correction as **TBXin** wrote above, or another TCR. Boxes are very different in schematics, and to achieve full compliance is difficult, especially if consider the quality.

RicKro, 2017-01-24:
I now tried with Shunt 96 % which seems to be quite good. I will also try to use TCR with higher values.

Thank you very much for your effective and quick help! Maybe It could be good to put a line in the readme related to the RX200S boxes for TC Mode?!

Have a great day!

maelstrom2001, 2017-01-24:
> RX200S boxes for TC Mode

Unfortunately, that will not be enough, the boxes are too different. It may be useful to note that we are using not high precise electronics and boxes with 1, 2, 3 and 4 cells, and what working on some model may not work properly on another. But I think that people who use custom firmware understand it and make allowances :)

closed 2017-01-24 by maelstrom2001