maelstrom2001/ArcticFox #82: Atomizer not found/Atomizer low

andreidotnet, 2017-01-24:
After playing around with the settings, I think I managed to break by RX2/3 with ArticFox.
Last setting I remember changing is Battery type from Generic to Samsung 25R, or TCR Setup to enable PI-Regulator (just to try and see what it does).
After this, the Coil resistance is not being read and shows 0. While firing it shows no atomizor found. I have reset the settings, re-flashed the ArcticFox firmware but still same behavior.
If i manually set the resistance to 0.25, when firing it shows atomizor low. This happens with multiple atomizors that work on other mods.
I have attempted to flash every stock RX2/3 firmware but it has the same behaviour - No Atomizor found.
Any ideas what might be causing this ? Have you guys encountered this issue while playing around with the firmware ?

[image]
[image]
[image]
[image]
[image]

maelstrom2001, 2017-01-24:
> stock RX2/3 firmware but it has the same behaviour - No Atomizor found

Check connector (positive post), try to pull it with something, tweezers for example.

andreidotnet, 2017-01-24:
So I guess there should not be any data that should survive between firmware flashes. I even used the reset dataflash option.
Thanks for the hint, will keep trying to pull the 510 connector. 
Strange coincidence.

maelstrom2001, 2017-01-24:
For better result disassemble box and make full revision: connector insulator, wires, soldering etc.
This is just a coincidence, nothing else.

traviswparker, 2017-01-24:
I'm not sure this is a coincidence. 

My RX2/3 died a couple of days ago in the _exact_ same way, two days after flashing the 20170120 firmware. I started getting TCR error and live resistance values that made no sense, and then just No Atomizer Found. At that point I could definitely smell a burned component.

Disassembled the mod, the wiring and 510 check fine with a multimeter... definitely not shorted, but the mod thinks the atomizer is open circuit even with a screwdriver placed across it. A dead short should say "Atomizer Low" not "No Atomizer Found" 

Same behavior with stock firmware, the hardware is definitely damaged. I'll put the board under a magnifier when i have the time and see if I can identify what failed.

andreidotnet, 2017-01-25:
Thanks a lot for the confirmation on your situation also. I checked my mod as well and cannot find any signs of failed wiring, but my electrical skills suck.
I will be sending into a local service shop, hopefully they can confirm from where the issue is.

@traviswparker: Please keep us posted if you find anything on your device. It would be nice to be able to confirm a possible compatibility issues/bricking issue for RX2/3.

maelstrom2001, 2017-01-25:
Found some info on Russian forum, maybe will be useful.
[image]
Failed (strikethrough) part is MOSFET driver, can be one of:
- [L6747A](http://www.alldatasheet.com/datasheet-pdf/pdf/577646/STMICROELECTRONICS/L6747A.html)
- [ADP3110A](http://www.alldatasheet.com/datasheet-pdf/pdf/281089/ONSEMI/ADP3110A.html)

Also there is info that "747A" part is less reliable than "L3E". Seems that RX2/3 owners faced the same hardware failure on stock Wismec firmware (even right out of the box), it can be easily googled too.
Anyway, have nothing to say is failure related to firmware or it is manufacturing defect.
All wrote above is **_just for your information_**.

traviswparker, 2017-01-25:
Yeah, I'd believe it is simply a shoddy part, especially if mods were coming DOA. I didn't think ArcticFox would do anything different as far as driving the MOSFETs, and my RX200 works fine on it on the latest builds. Let's hope the Cuboid I'm replacing it with is well built.

On that note, do you still enforce the 25A power output limit on the Cuboid, or is that bypassed when the duration is limited (preheat, temp ramp-up, etc.) that would be fantastic if so!

maelstrom2001, 2017-01-25:
> Let's hope the Cuboid I'm replacing it with is well built.

Wish you luck! Cuboid has another weak spots: batteries charger and resistance measuring. Beleive that mass defects were in early production phase, and now all fixed.
Yes, the limit in 25 amps is still intact. Perhaps this is reassurance, but we prefer do not modify critical limits set by manufacturer.

traviswparker, 2017-01-26:
Meh, the 25 Amp output limit seems to not matter at all with Ni200 because the resistance increases so quickly. 150W+ a fraction of a second when firing a 0.13 build. Never even hits 25 Amps.
[image]

I don't think the stock FW recalculates max available power based on live resistance, because I get a _very_ weak ramp-up on low builds.

Sanaki, 2017-01-27:
While this seems to be established already, I'd like it known that ArcticFox doesn't cause issues universally with the RX2/3. I've had no issues whatsoever with mine that I didn't produce through my own mistakes.

heavydetail, 2017-02-08:
hi,
my rx2/3 now also says "atomizer low" even when you short the 510 and it won't fire anymore. i've been updating from 4.02 stock fw to the latest AF release. i thought i could repair my display problem as stated here in this way: 

https://github.com/maelstrom2001/ArcticFox/issues/117

https://www.reddit.com/r/electronic_cigarette/comments/5simbq/help_wismec_rueleaux_rx23_dead_screen_epidemic/

bummer...didn't think this would make it worse than it was before. :( now i can't vape at all, before only the display was effed.

TBXin, 2017-02-08:
Read here: https://github.com/maelstrom2001/ArcticFox/issues/117

Our firmware can't harm your device.

heavydetail, 2017-02-09:
Before Flashing: screen didnt work, but it fired,
After Flashing: it didnt fire anymore and screen still didnt work.
The harm: Mod is now unusable.

What the cause was and what your definition of harm is, idk,
but the situation got worse by flashing AF, so I don't recommend it for people with display problems.
Im not blaming you, it was my choice, but saying your firmware can't harm the device, even if it was already faulty, is wrong. qed.

Mine went back for warranty today, will try again in a few weeks.
Its a great firmware otherwise.

closed 2017-02-12 by TBXin