maelstrom2001/ArcticFox #83: Is the code opensource?

andreivoinea93, 2017-01-24:
Hello,

First of all I want to thank you for your effort, you're really putting new life in our devices. Your project is a really wonderful alternative to DNA devices and Evolv's software.

I have some knowledge regarding microcontrollers, Atmels and Microchips mostly and I'm willing to help your project in my spare time, if I will be able to make someone quit smoking and save a life with a pull request, why not? I can't find the source code for the firmware anywhere on this github project. Is it opensource? If so, where can I find it?

Have a great day!

maelstrom2001, 2017-01-24:
Thanks! Currently firmware is distributed "as is", no source code provided.

andreivoinea93, 2017-01-24:
Are there any plans on doing so?

maelstrom2001, 2017-01-24:
At this point no. Maybe later, when we'll stop development and support.

closed 2017-01-24 by maelstrom2001