maelstrom2001/ArcticFox #88: [Suggestion] SmallScreen-Layout in TC Mdes

stardealer, 2017-01-26:
Is it possible to make "Non dominant (Pwr/Temp)" available in SmallScreen-Layout as it is in ClassicScreen?
Same question goes for CircleScreen.

maelstrom2001, 2017-01-26:
Circle skin will not be modified. It's our concept and our point of view :)
96x16 layout - maybe.

traviswparker, 2017-01-26:
I thought the same thing but then I realized if you put the non-dominant setting on the small screen, when you enter edit mode it would be displayed twice and only one of them can be changed. Would be very confusing. It's easy enough to assign 2-click to t.dom and quickly switch back and forth.

stardealer, 2017-01-26:
I don't think it would cause much confusion. Not so much different from today because if you choose for example to display "last temperature" you see two temperatures and only one can be changed. So confusion potentially exists already or there is and will be not so much confusion at all.

I would like to see the two main parameters of TC at a glance and not by clicking back and forth.