maelstrom2001/ArcticFox #89: [Only FYI]

Obicom, 2017-01-26:
Second part of my YouTube review about ArticFox is released. (German Language)
In this part I compare the stock firmware with AF with different measurements and explain some functions a little bit more in detail. (Clock setup, Pre-Heat delay, BVO, PI-Control)
I hope you will enjoy the video ;)

https://www.youtube.com/watch?v=XXh3PrihtoU

maelstrom2001, 2017-01-26:
Great review! :+1: Much appreciated, thank you very much!

smok-e, 2017-01-26:
Super Obi, great video, the german translation should now be ready
without any mistakes. But I still think if it should betranslated as "X32 Kristal" or "X32 Quarz",
both are possible, but which  is better?

vapinggirl, 2017-02-01:
Good work Dirk and many thanks for your comparision between Stock and AG fws 

Thats why i decided to Order a rx300 :)

In this case imho Quarz is the correct translation

closed 2017-03-29 by maelstrom2001