maelstrom2001/ArcticFox #91: [Question] More information about PI control

Obicom, 2017-01-27:
@maelstrom2001 : Is a souce available with more concrete information about the PI Temp control settings? What does the "Range" exactly mean? In which range the P and I value are working? What make sense what not? Do Pre-Heat work in temp mode? If I observe the display I would say no ....

TBXin, 2017-01-27:
@Obicom 

PI = proportional–integral controller, you can find tons of information about this kind of thins in the internet. Like this one: https://en.wikipedia.org/wiki/PID_controller

Range is a point, at which PI regulator starts working. Percents is how close we are to a predetermined temperature. 0% - PI Regulator start working right after you press fire button. 100% - PI Regulator start working when you coil temperature is equal or higher than setted.

P: min: 0, max 6000
I: min 0, max 9000

About preheat and PI-Regulator. PI-Regulator have a higher priority than preheat, so if it determine that temperature is growing in the not expected way - it will try to reduce it, ignoring preheat values.

To make preheat works with PI-Regulator you should move range value further from the beginning... 30-40%

Sanaki, 2017-01-27:
What is the base value for 0%? While in practice I'm sure an explicit 0% would simply trigger at all times, 50% of 350F is undoubtedly not 175F, nor 250F. I would imagine 70F would be considered a "standard" baseline temperature, but knowing the exact value would still be beneficial.

sandbagjohnny, 2017-01-27:
Thanks for this. To verify,(apologies if bad question). the proportional value (P) is a gain for.. which one, voltage? or wattage?

..and is (I) time in milliseconds?

Thanks. Awesome software package!

maelstrom2001, 2017-01-27:
@Sanaki 
> What is the base value for 0%?

If you set 350F and 50% Range in profile, PI-regulator will be switched on when coil temperature exceed 175F. You will not get power reducing immediately at this temperature, but from that moment PI-regulator will control power according to temperature.

@sandbagjohnny 
**P** = some_value * mW / C (temperature error)
**I** = some_value * mW / C / s (temperature and time)

Sanaki, 2017-01-27:
> If you set 350F and 50% Range in profile, PI-regulator will be switched on when coil temperature exceed 175F.

So the base value for the percentage is 0F or 0C depending on which temperature scale you currently have selected?

maelstrom2001, 2017-01-27:
@Sanaki apologies for mistake, my bad. Have to read source code to recall...
As PI-regulator based on Celsius, the base is 0C always.
So, in your example 100% means 350F = 177C, 50% = 177 / 2 = 88,5C = 191F
I know, this is not obvious, maybe because we don't use Fahrenheit degrees. On other side, this is not reference value, it can be set very approximately if you plan to use TC with power curves, preheat etc.

Sanaki, 2017-01-27:
> The base is 0C always

Not a problem, that's the exact info I was looking for. Thanks.

sandbagjohnny, 2017-01-28:
@maelstrom2001 Thank you for this info!

sandbagjohnny, 2017-01-30:
@maelstrom2001 Thank you for this info! [I've posted an essay/mini-review of the new temp control methods on reddit](https://www.reddit.com/r/electronic_cigarette/comments/5ql2kn/on_pi_regulation_and_arcticfoxmyevic_some_notes/) based on both your comments here, and [clockselect's comments on PI in his fork](https://github.com/ClockSelect/myevic/issues/132)

maelstrom2001, 2017-01-30:
@sandbagjohnny thank you! :)
> The default P value appears to be set up for a heavy (clapton, etc) coil

...or 2 simple SS coils. Yes, on box side this simple setup looks like something heavy due to low TCR and resistance inequality.

> if you're using plain wire the p value can be much lower.

True. This is PI-regulation with default presets (Range = 0%, P = 1850, I = 300) on single titanium coil (28AWG, 5 wraps, genisis atty - "lightweight"):
[image]

And when I set P = 800, I got this:
[image]

sandbagjohnny, 2017-01-30:
Thanks for the comparison! I've updated the mini-review.

closed 2017-02-28 by TBXin