maelstrom2001/ArcticFox #92: Upload from NFE Toolbox redetects atomizer, reassigns profile

Sanaki, 2017-01-27:
When uploading a configuration from NFE Toolbox, the current atomizer is treated as though it was just attached. While in normal operation this isn't an issue, I run titanium coils and use smart profile swapping. As a result, if I've been tweaking settings and testing between uploads, my resistance changes enough (about 65% increase from cold to hot) to require reassignment of a profile.

From my end, this requires one of a few things. I can disable smart profiles (not preferred), manually change my locked resistance after reassignment (time consuming), or wait around a full minute with my cap removed for the coils to cool down fully before uploading. The only simple solution I can think of is to provide a way to choose a profile without overwriting the resistance value stored in it. I would imagine treating the attached atomizer as freshly attached after an upload is a preferable behavior, but if not, that would do the trick too.

Not sure exactly how to categorize this one or what the best "fix" would be, as it seems like a positive design choice with an unintended consequence.

maelstrom2001, 2017-01-27:
> When uploading a configuration from NFE Toolbox, the current atomizer is treated as though it was just attached.

By design. Maybe later will did something with it, but not at this point.

Sanaki, 2017-01-27:
> By design.

That's what I figured, but at least now it's a known side-effect.

maelstrom2001, 2017-01-27:
We know it.

closed 2017-01-27 by maelstrom2001