maelstrom2001/ArcticFox #93: Please add power bank mode like rs300x!

userdsp, 2017-01-27:
Please add power bank mode like rs300x!

Wietjuh, 2017-01-30:
would love to see this... especially for on festivals/on the road

docdeath, 2017-01-31:
really i own rx300 my self and no way would i trust it as power bank  to. its just gimmick  added as selling point.  i got mine for battery life for vaping and nothing more

wulfgrim, 2017-02-18:
This right here... I'm not sure it'd be worth the effort it would take to make it happen, It would be cool as hell. but you're looking at taking a device that's designed and programmed to drive a restive load and telling it to drive a LC load. there could be issues.

the rx300 might have extra hardware to make this possible.

while the device can drive a TC load, that would be similar to having it hold  a steady 5.1V on it's output. I dunno....

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001