maelstrom2001/ArcticFox #96: [Question] Battery discharge profiles - Info needed

FrankBKK, 2017-01-29:
Can someone point me to more information on how to create discharge profiles for individual battery types ? 
Where and/or how to find the proper data sheets etc.

smok-e, 2017-01-29:
which batteries? data sheets might be somewhere on the i-net, have you tried google?

smok-e, 2017-01-29:
and you can search for mooch the battery guru ;-)

maelstrom2001, 2017-01-29:
@FrankBKK you can use discharge profiles for DNA, for example. But the most precise result you can get if you did it yourself. Charge battery full, reset statistics, vape and fill table like that:
[image]
It contains pairs of numbers, total vaping time and battery voltage. Then make chart and draw the same in NToolbox.

FrankBKK, 2017-01-30:
Perfect - Thanks for the Info !

closed 2017-01-30 by FrankBKK